bastibl/gr-ieee802-11 #279: HackRF not Transmitting when using wifi_tx.grc file

antittelbaugh, 2021-06-08:
I am currently trying to use the wifi_tx.grc and the wifi_rx.grc to transmit and receive the 802.11a wifi signal on two distict HackRF's running on separate computers. To do this I replaced the UDH blocks in both functions with osmocom source and sink blocks (added from gr-osmosdr) and adjusted the settings accordingly. The wifi_rx.grc does work as it was tested with a 802.11a signal  generated through other technology, however it nor the spectrogram in my lab are receiving a signal when I run the wifi_tx.grc function with the hackrf connected.  I am running GNU radio 3.7.11. 

To try to fix this I have changed between antennas between an ultra wide band and an RF Explorer H-Loop Near Field. I have added Frequency and time plots to the wifi_tx.grc which appear as they should. I have dectected a small change in the magnetic field near the antenna which tells me it doing something. Finally, I ran it with a LimeSDR at 2.4GHz which appeared to work (the osmocom sink block was replaced with a Lime sink block) but I need to be able to transmit to 5GHz which the lime is not capable of. 

Some errors I have noticed
The wifi_tx.grc function consistently prints underruns when running the HackRF( though not on the Lime), I have run volk_profile in an attempt to fix this with no avail.

Initally I thought that sense the lime is USB 3 compatible while the HackRF was only USB 2 compatible the HackRF could simply not keep up, however in reading the paper "Performance Assessment of IEEE 802.11p with an Open Source SDR-based Prototype" I see that this code was used with the HackRF sucessfully.

Any suggestions or help would be appreciated

bastibl, 2021-06-09:
Some ideas:

- make sure that you only changed the Sink and didn't change/remove any other blocks.
- check the supported gain stages and set a higher gain
- use other GNU Radio blocks (e.g., signal source), send any signal, and check if you see it in the spectrum (also try the same center frequency and bandwidth)

If the latter doesn't work, that's an issue w/ the HackRF, the HackRF driver, gr-osmosdr, or GNU Radio.

With the HackRF, there's probably an underrun at the end of every frame. So as long as the reported underrruns correspond roughly to the frame rate, that's fine. But even if there are more, there should still be a signal (It might be corrupted and not decodeable, but there should still be a signal.)

antittelbaugh, 2021-06-10:
Thank you!
I have double checked and I have only changed the sink.
I have changed the gain to no avail.
I have set up a simple source signal -> rational resampler -> sink diagram to transmit and while it does as you get past 3GHz it gets harder and harder to differentiate from the noise floor such that it cannot be seen past that point. 
Is this a problem you've seen before? 
Thank you for your help

bastibl, 2021-06-12:
No, but I don't have much experience with the HackRF, so I'm not very qualified to help in that regard. You could ask in the GNU Radio Matrix channel, GNU Radio mailing list, or open an issue in the HackRF repo.

closed 2021-06-12 by bastibl