bastibl/gr-ieee802-11 #282: no packet captured in Wireshark

ramziB99, 2021-06-14:
Hello, 
Im using wifi_rx.grc to capture wifi packet with LimeSDR USB in a raspberry pi 4. 

I modified the flow graph to use my sdr device by replacing UHD block with LimeSDR source (RX), when I run rx_demo.sh in order to see if I capture any packet I I have none in the Wireshark window. I tried several freq and sample rates but nothing changes. 

Do you have any idea of what's wrong ? 

Best, 
Ramzi

bastibl, 2021-06-14:
Make sure that there are no (or only few) overruns. If the LimeSDR has an uncompensated DC offset, try removing it with a filter. Make sure that whatever you try to receive is 11g without any compatibility modes (in 5GHz, this is less problematic. So use 5GHz if you can). You could also use another SDR as transmitter to make sure you are trying to receive frames in the correct format. Try different gains and check the antenna.

ramziB99, 2021-06-15:
hello, 
I tried several things you propose to me but still not working but I suspect something. Since I'm a using a LimeSDR USB I replace the uhd block by LimeSDR source RX, I put the frequency to "freq" so maybe when I start the flow graph it's always the same frequency. How can I translate this line from uhd block "uhd.tune_request(freq, rf_freq = freq - lo_offset, rf_freq_policy=uhd.tune_request.POLICY_MANUAL)" to my LimeSDR source block.

bastibl, 2021-06-15:
Just putting `freq` into the field for "RF frequency" is fine. This complicated looking tune request is for offset tuning, which is not supported by LimeSDR (at least as far as I know). I'd assume that this is not the problem. To make sure, you can add a frequency sink directly after the LimeSDR source and check if you see your transmissions.

ramziB99, 2021-06-17:
I just tried to do the same thing with hackrf one and it worked but nothing with the LimeSDR USB, that's really weird

bastibl, 2021-06-18:
Instrumentation -> QT -> QT GUI Frequency Sink. You can search for blocks wtih CTRL-F. If you have issues with the SDR you could also ask at Lime. I don't have such a device.

ramziB99, 2021-06-22:
Still have the same problem, the only difference I see between LimeSDR and the HackRF is in the Source block. For the HackRF we use osmocome source were I can put 'freq' for the frequency while for LimeSDR source I can't because the device don't support 5ghz frequency so if I want it to work I have to put a frequency like 2.437 Ghz. Im not sure if this could be the problem but im running out of idea.

ramziB99, 2021-06-23:
I resolved the issue, I had to change the default option of freq which was set to 5.89g which is not compatible with my limeSDR usb. After changing the default option and setting the frequency to "freq" in the LimeSDR rx source everything worked fine.
Thank you for your help, I will close the issue.

closed 2021-06-23 by ramziB99

jackdvd, 2022-07-03:
Hi @ramziB99, I have similar issue with LimeSDR and it still doesn't work for me. Did you use any additional configuration for the channel A? For calibration, analog filter and digital filter bandwidth?

jackdvd, 2022-07-04:
Never mind, I have figured it out. I used the sample rate for analog filter and 0 for digital filter and it worked.