bastibl/gr-ieee802-11 #284: Is it possible to use this module in Windows or WSL?

Kevinous, 2021-06-19:
As the title, can I  use this module in Win10 or WSL?  For example, build from source in Linux and copy the blocks to some folder in Win10?

bastibl, 2021-06-21:
I don't have experience with Windows and am not sure about the current state and support of GNU Radio on Windows. This module is not special. I'd recommend to ask on the GNU Radio mailing list or Matrix chat about Windows and OOTs.

closed 2021-06-21 by bastibl

Kevinous, 2021-06-21:
The only reason that I didn't use Linux VM is that the computation power of VM cannot support a high sample rate like 500k. So I have to try to use GNUradio in my original system... Thx for your information though!

bastibl, 2021-06-21:
If you allow the VM to access (multiple cores, some RAM, etc),  500k should be trivial to achieve. Some reported performance issues of USB-based SDR (even though I was not able to reproduce this issue). Anyway, I think using it natively is more fun.
Here is some more information, if you at least want to give it a try on Windows: https://wiki.gnuradio.org/index.php/WindowsInstall#Windows_Installation
But I'm not sure what option works best.

Kevinous, 2021-06-21:
Thanks for your help! Actually I have installed GNURadio 3.7 on my Windows and I have run some flow graph with my USRP N210. I just wonder how to install OOT module since many of them do not mention how to install with Windows...

Kevinous, 2021-06-22:
Btw, can I compile and install this module with Conda? Since I saw some conda can be used in Windows to install some module..

bastibl, 2021-06-22:
AFAIK, someone put a lot of time in supporting GNU Radio (and OOTs) with Conda. But, again, this is not specific to this module and depends on GNU Radio support of these systems. Please ask on the GNU Radio mailing list or the Matrix chat.

Kevinous, 2021-06-22:
Got it. Thanks for your patience!

FlyerMaxwell, 2023-03-16:
> Thanks for your help! Actually I have installed GNURadio 3.7 on my Windows and I have run some flow graph with my USRP N210. I just wonder how to install OOT module since many of them do not mention how to install with Windows...

Hi, How you configure your WSL to connect USRP? When I use WSL2, I cannot find USRP with "uhd_find_devices".