bastibl/gr-ieee802-11 #287: Signal strength while using wifi_rx

ramziB99, 2021-06-23:
Hello bastian, 
I would like to know how the Signal strength is computed ? for the experience I want to do, I need the most accurate signal strength  possible so I need to know where this value come from and how can I add grc blocks to have the wanted accuracy if it's not the case.
[image]

Best, 
ramzi

bastibl, 2021-06-23:
The SDR is no calibrated measurement device. You'd need to calibrate for bandwidth, gain, and frequency and then map average magnitude to a power/signal strength.

The receiver only estimates the SNR and sets noise and signal level so that the SNR makes sense. In your case, the estimated SNR was 29dB.

ramziB99, 2021-07-01:
Do i have to make changes in the flowgraph of wifi_rx to do that ? sorry if my question seems dum, I am an intern in a research lab and this field is pretty new for me. 

I also have another question, is it possible to capture packets from a specific Mac address ? 

thank you,

bastibl, 2021-07-01:
Power calibration is something that is independent from this module. UHD recently introduced a power API, but I didn't have a look yet. Please ask on the GNU Radio or UHD mailing lists or the GNU Radio Matrix chat.

Filtering frames based on the MAC address is not done on the PHY but at a later stage. For example in Wireshark.

closed 2021-07-13 by bastibl