bastibl/gr-ieee802-11 #289: Trouble with installation

didahalpha, 2021-06-28:
Sir,

Excuse this, probably repetitive "issue", but I have literally spent more than 30+ cumulative hours, attempting to install GNUradio, the GR-osmosdr, this module, as well as your gr-foo module. At no time, during installation processes, both by using "apt install gnuradio", and or compiling your gr code and GNUradio from (Github), I am unable to get all blocks of this code to load correctly to be used, while using Ubuntu 20.04.

I have read the other "issues" that have been reported to you, over and over, and only found one repetitive answer by you; all files must be of the same version / type (Master / Maint). However, following that answer, I thought I had done everything correctly, only to find out that I didn't; eventually not even getting osmosdr to work, which I had previously done, and brought in a live signal. 

Just FYI, and again, I have been successful in the past, getting most everything loaded where the graphs show up, but some connections, specifically the Gr-foo processes, are red'd out, and can't be used, because The Gr-foo module cant be located.
Now, it seems like my entire load (after multiple attempts to install), doesn't work at all, and I'll have to wipe my computer clean, and start over, which is okay.  

My question to you Bastion, if you would please take the time to answer, - because I'm highly interested in learning how to use GNUradio and its functions - could you or someone reading this, give me step by step directions (or show me, in order where they are), an order of installation of each module, where each module goes onto the computer, what variables and Paths to set, and specifically where to get each file (github / apt install)? Even at 49yo, I am extremely lost in this new venture, and could use a professional's help. All of the instructions that I have found, in multiple places (webpages), can be very daunting to a newbie.
Thank you sir / everyone.

bastibl, 2021-06-30:
I understand that this must be frustrating, but I'm not sure what exactly you are expecting. You don't ask for help for a specific problem but seem to expect a complete tutorial that covers everything. I'm afraid that's more than I can provide at the moment. I also use Ubuntu 20.04 on all my systems and never had an issue. On high level, I'd recommend to:

- Install GNU Radio 3.8 (`gnuradio-dev`) and Swig through the package manager (do not use the GNU Radio 3.9 PPA)
- Install the `gr3.8` branch of gr-osmosdr
- Install the `maint-3.8` branches of gr-foo and this module

If there are any issues with this process, please open an issue in the corresponding repository about the specific problem you face. Since you seemed to test various options, please make sure to uninstall existing aritfacts beforehand, otherwise this might lead to very strange problems that can be hard to debug.

didahalpha, 2021-06-30:
Sir,
The steps that you have provided may be the answer I'm looking for, and will give it a shot as soon as I have access to my computer, and can do a complete operating system reload.
I do appreciate your time, and if this works, I'll drop a note to let you know.
Thanks again Bastion.
Brad

closed 2021-07-13 by bastibl