bastibl/gr-ieee802-11 #293: Ask the question of the data saving

lms7657061, 2021-07-08:
hello, sorry to bother you again, and sorry for my poor Enflish. I want to ask that when the wifi data is saving in the file sink and the file wifi.pcap, whether the data is consist of the preamble?

bastibl, 2021-07-08:
The PCAP file is supposed to be opened with Wireshark. It captures only higher-level data, similar to a WLAN card in monitor mode.

lms7657061, 2021-07-08:
I want to get the preamble of the frame, So how I can get the preamble of the OFDM frame?

bastibl, 2021-07-08:
If you want the full preamble (complete long and short), you'll have to create a block that is connected to the output of sync short that dumps a given number of samples after each `wifi_start` tag. If the full symbols of the long training sequence is enough, you can log them in the frame equalizer.
In any way, there is no config option or similar for this. You'll have to create a trivial, custom block.

closed 2021-07-13 by bastibl