bastibl/gr-ieee802-11 #295: HackRF receiving WiFi beacon (gnuradio 3.7)

jamesshao8, 2021-07-13:
Hi bastibl,

I have a computer running ubuntu 16.04 and gnuraido 3.7.11. I can successfully receive beacon using limesdr and osmocom. But can't do that with HackRF. I am figuring out the reason.

I have read in this issue.
https://github.com/bastibl/gr-ieee802-11/issues/248
It says that if using hackrf and osmocom, it is necessary to upgrade gnuradio to 3.8? Can you explain what reason is behind this?

Thank you.

bastibl, 2021-07-13:
This is a very convoluted issue. You don't have to upgrade. Please search for HackRF in the issues. The topic came up very often. In a nutshell, you'll have to remove the DC offset through a filter.

jamesshao8, 2021-07-13:
> This is a very convoluted issue. You don't have to upgrade. Please search for HackRF in the issues. The topic came up very often. In a nutshell, you'll have to remove the DC offset through a filter.

Thanks for your reply. I checked the issues about this topic before. But I didn't find a flowgraph about that. I suppose the filter to remove DC offset is something like removing average value of iq stream, right?

Would there be also other problem of HackRF? For example its clock isn't as precise as what's in LimeSDR. It could cause problem for GPS signal but I am not sure how important it is for WiFi?

bastibl, 2021-07-13:
I linked an adapted flowgraph here:
https://github.com/bastibl/gr-ieee802-11/issues/252#issuecomment-774479845
And here:
https://github.com/bastibl/gr-ieee802-11/issues/243#issuecomment-736295258
And here:
https://github.com/bastibl/gr-ieee802-11/issues/83#issuecomment-369575692
And probably in many other places,

closed 2021-07-13 by bastibl

jamesshao8, 2021-07-14:
Thanks a lot.
I had error when opening your flowgraph. But somehow I managed to copy the FFT filter out, and finally succeeded running on my side gnuradio(3.7.11). I hereby attach my flowgraph in case anyone need it in future.

[wifi_rx_hackrf.zip](https://github.com/bastibl/gr-ieee802-11/files/6813359/wifi_rx_hackrf.zip)
[image]
[image]
[image]