bastibl/gr-ieee802-11 #296: How to get the Scrambling Seed

lms7657061, 2021-07-13:
Sorry to bother you again! I have some question of the scrambling seed, the code in the below:
void descramble (uint8_t *decoded_bits) {

	int state = 0;
	std::memset(out_bytes, 0, d_frame.psdu_size+2);

	for(int i = 0; i < 7; i++) {
		if(decoded_bits[i]) {
			state |= 1 << (6 - i);

		}
       

	}
	out_bytes[0] = state;
whether we can recover the transmitter’s 7-bit linear feedback shift register by first computing state bit values? if not, how to computing? I am not very unstand the descramble code.
Thank you!

bastibl, 2021-07-13:
I don't understand, but the scrambling seed corresponds to the `state` variable.

lms7657061, 2021-07-16:
Ok，Thank you!  I have another question, if I want to only get the management frame, How should I do? How to modify the code?

bastibl, 2021-07-17:
You can just filter in Wireshark. Or you have to parse the payload yourself, similar to the Parse MAC block.

https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/parse_mac.cc#L73-L78

lms7657061, 2021-07-19:
well, thank you very much! I notice that the beacon frame there have the SSID and it can print in the terminal, but the probe request frame don't print, I konw sometime the SSID of the probe request is null, but if I assign a SSID to the probe request, how I should modify the code.
case 4:
			//dout << "Probe Request";
                        dout << "Probe Request" << std::endl;
			if(length < 38) {
				return;
			}
			{
			uint8_t* len = (uint8_t*) (buf + 24+1);
			if(length < 38 + *len) {
				return;
			}
			std::string s(buf + 24 + 14, *len);
			dout << "SSID: " << s;
			}
			break;
		case 5:
			dout << "Probe Response";
			break;
		case 6:
			dout << "Timing Advertisement";
			break;
		case 7:
			dout << "Reserved";
			break;
		case 8:
			dout << "Beacon" << std::endl;
			if(length < 38) {
				return;
			}
			{
			uint8_t* len = (uint8_t*) (buf + 24 + 13);
			if(length < 38 + *len) {
				return;
			}
			std::string s(buf + 24 + 14, *len);
			dout << "SSID: " << s;
			}
			break;
Case 4 is modified by reference to Case 8 ,but didn't print the correct SSID, so how I should modify the code? please help! thank you！

bastibl, 2021-07-19:
This block was never meant to be a full-fledged WiFi parser but some trivial tool to help testing that the decoder works. If you want to extend it, you have several options:
- study the frame format, looking at the standard
- parse the payload with Scapy
- open the PCAP file with Wireshark

I'd recommend to enable the Wireshark Connecter block and open the PCAP file in Wireshark.

closed 2021-09-06 by bastibl