bastibl/gr-ieee802-11 #301: transmit the beacon frame

lms7657061, 2021-07-22:
hello! well, I've been studying your gr-ieee802.11 project lately and konw use Scapy to broadcast forged beacon frames for a fake Wireless Access Point. So I want to konw whether we can make the USRP as the true AP and send the beacon frame, then the smartphoneconnects to the AP and communicates with it?

bastibl, 2021-07-23:
Unfortunately this i not possible, since this would require a complete stack, including MAC, that meets the timing requirements of the standard. This is not possible w/ a PC implementation. You'd have to look into OpenWifi for that.

lms7657061, 2021-07-23:
Ok, I get it, thank you! but I have another question: when I use the scapy and send the beacon frame as https://archive.fosdem.org/2019/schedule/event/gr_scapy/attachments/slides/3366/export/events/attachments/gr_scapy/slides/3366/gnuradio_meets_scapy.pdf. I used frequency 2457MHz, bandwidth 20MHz, tried with BPSK QPSK QAM but my iphone can't found the ssid called "GR WLAN" , however I can see it in the wireshark,  I don't konw why, can you tell me?

bastibl, 2021-07-29:
Debug it step by step. I assume you only see the frames in a local Wireshark dump. So first make sure that other devices can receive the frames from the example TX flowgraph, i.e., use another SDR or a PC with a WLAN card in monitor mode as a receiver. If this works, use Scapy to send the beacon frame. Again make sure that an SDR or PC can receive it. If this works but the iPhone doesn't, try to increase the rate at which frames are sent and try different gain settings.

closed 2021-09-06 by bastibl

Windy-with-sky, 2023-04-18:
Hello，can I make openwifi connected with usrp

bastibl, 2023-04-18:
This is not the openwifi repo. And the fact that this work w/ USRPs is the first sentence of the readme.