bastibl/gr-ieee802-11 #302: the question of frequency offset

lms7657061, 2021-08-03:
Hi, sorry to ask the question of frequency offset again, but I have the question of sync_long.cc. whether the code "pmt::from_double(d_freq_offset_short - d_freq_offset)"  annotates the overall frequency offset to downstream blocks? if it is, why not "pmt::from_double(d_freq_offset_short - d_freq_offset)"? because I learned from other sources that the overall frequency offset is equal to the course frequency offset plus the fine frequency offset? why in your code it is d_freq_offset_short - d_freq_offset?

bastibl, 2021-08-05:
You posted two times the same code. But yes, it's the overall frequency offset. If you add or subtract is a details that depends on how you calculate it. For example if you do `sum(sample[i] * conj(sample[i-64]))` or `sum(sample[i-64] * conj(sample[i])`

lms7657061, 2021-08-06:
Oh, thank you very much! I have got it!

closed 2021-08-06 by bastibl