bastibl/gr-ieee802-11 #305: About WiFi packet transmission time

Leoplt, 2021-08-11:
Hello, In example wifi_tx.grc  ,  I want to know whether the duration of WiFi packets during transmission can be controlled through existing parameters (e.g. samp_rate, pdu_length, interval)?
Thank you for your attention and look forward to your reply！！

bastibl, 2021-08-16:
You can increase the PDU lenght, i.e., add more payload to make the frame longer. Also changing the Encoding makes the frame longer (BPSK 1/2 results in the longest frames).

Leoplt, 2021-08-17:
Thank you for your reply!
I want to get the transmission time of each WiFi frame.
How to calculate the transmission duration of each WiFi frame？

bastibl, 2021-08-23:
It's non-trivial calculation that I'd also had to look-up in the standard. You could create the frame with this module and see in how many samples it results but that's a bit overkill. In general it it is: 2 symbols short preamble + 2 symbols long preamble + 1 symbol signal + X symbols for the payload, which depends on the MCS.

closed 2021-11-24 by bastibl