bastibl/gr-ieee802-11 #307: 10 ms interval remove

alisabra94, 2021-09-03:
Hello,

Is there anyway I can remove the 10 ms interval so that the Tx transmits contineously ?

Kindly,
Ali

bastibl, 2021-09-06:
There is a PacketPad block that adds zeros around a frame. You can remove the block or lower the padding, but if you generate the frames on the fly, the actually issue is probably that the transmitter is not generating the frames fast enough. This can be overcome by generating frames with the TX flowgraph and write the samples to a file. Then, just send the file (potentially in a loop).

alisabra94, 2021-09-22:
Thank you so much for your quick response.
Can I use the File sink block write the sample ? Then just stream the same file later in a loop using the File source ?

Kindly
Ali

alisabra94, 2021-09-22:
Correction:  to write the samples **

bastibl, 2021-09-23:
Yes, exactly. You modify the TX flowgraph to write the samples to a file. And modify the RX flowgraph to read from that file. The file source has an option to loop the file.