bastibl/gr-ieee802-11 #311: HackRF one Transmitting/Receiving ITS-G5 using wifi_rx/tx

YamanDaif, 2021-10-04:
Hi Bastian! I'm using [wifi_rx.grc](https://github.com/bastibl/instant-gnuradio/blob/wlpc/assets/wifi_rx.grc) flowgraph you uploaded for the hackrf.
When trying to receiving test packages from Cohda wireless mk2 the output is "UUUUUUU...." and the file wifi.pcap is only 24byte and empty when i open it with Wireshark. As I understood from earlier issues you assisted the UUUU output might be underflow, (one of the blocks is trying to revive data faster than it is being provided) but with my little experience with GNU radio, I can't really spot the blocks that might cause this issue..
I also read from earlier issues you assisted that this might be an off-set but since I'm using the exact same flowgraph you uploaded for the hackrf is neglected this assumption.

When transmitting from the hacker, the Cohda wireless boxes doesn't detect anything and I'm not sure what is the problem here. For transmitting, I'm using the wifi_tx.grc file and changed only the sink to (Osmocom sink) for HackRF compatibility. 
 The help with tracking down the problem I'm having with my setup would be much appreciated! 
Attached below you can find the flowgraph from the wifi_rx I'm using with some 
We are using GNU Radio Companion 3.7.9 installed on Ubuntu 16.04 persistent live USB.

PS. wifi_loopback.grc works fine without any problems.

[image]

bastibl, 2021-10-06:
It is OK if there is one underrund (i.e., one "U" per frame) since the HackRF doesn't now that the frame ended and thinks it is an underrun. Also, you are using a sample rate of 1MHz, but IEEE 802.11p uses 10MHz.

bastibl, 2021-10-11:
There was no science behind my tests. The flowgraph you linked produces a frame every 300ms. If the Us appear in about the same rate, it's probably OK. Also if a receiver is able to receive all sent frames, there were no underruns during the frame, since this would have corrupted the transmission. I'd assume these type of tests are good enough.

closed 2021-10-12 by YamanDaif