bastibl/gr-ieee802-11 #316: Errors running `wifi_phy_heir`

arsenovic, 2021-11-09:
following the install instructions.  when i try to compile the `wifi_phy_heir` i get, 

[out.txt](https://github.com/bastibl/gr-ieee802-11/files/7506303/out.txt)

i also had to manually move the ieee802_11 python module to the right place ( i have conda so perhaps this confused the install make build ) 

any tips are appreciated thanks!

bastibl, 2021-11-09:
If you want to use this module with Conda, I'd recommend to package it for Conda. Since you had to manually copy stuff around, I'd assume that you missed copying something. Maybe the Python bindings cannot find the library, they are linked against. You could open an interactive Python session and see if this provides more information, e.g., can you import the module and it is just missing stuff C++ bindings, etc. Also check the cmake output and make sure that the Python bindings were generated in the first place (maybe siwg/pybind11 was missing).

arsenovic, 2021-11-09:
thanks for the response. i was able to run `python .grc_gnuradio/wifi_phy_heir.py`, so it appeared that the python bindings were installed correctly.  however, i have since tried to get pybombs working, (which failed, is missing some deps). i then found a vbox image https://wiki.gnuradio.org/index.php/UbuntuVM , 

for my needs i have saved binary iq files, and i was hopping to use this libarary to  demodulate the packets from these files.  not in realtime from a SDR, so perhaps this vmbox image will work?

arsenovic, 2021-11-09:
whats the easiest way to install this? the pybombs link does work, not sure if this is still the recomended way?

bastibl, 2021-11-09:
I'd recommend to install the GNU Radio dev package (usually gnuradio-dev) through the package manager of your distribution. Then, compile the corresponding version of this module (maint-3.8 for GNU Radio 3.8, maint-3.9 for GNU Radio 3.9).

It sounds like you have now a broken PyBombs and a broken Conda installation. Make sure that you environment variables do not point to them. Otherwise you'll get the weirdest errors that are hard to debug.

closed 2021-11-24 by bastibl