bastibl/gr-ieee802-11 #317: BladeRF with Transceiver.grc

YamanDaif, 2021-11-12:
Hi Bastian! 
I'm having trouble with getting the wifi_transceiver.grc work with bladeRF and I wonder if you have any advice I might use to make it work! 
I ready tested wifi_rx.grc by changing the source to gr-osmocom source and it seemed working. however, the Wireshark pcap file shows nothing (24 bytes) and I  don't know if it's a DC offset that I'm not aware of or if it's a Wireshark. 
the help to make it work would be much appreciated! 
This is a screenshot of the wifi_rx I'm using.

[image]

bastibl, 2021-11-12:
What are you trying to receive? There is no WLAN mode that uses 30MHz... You cannot just change parameters arbitrarily. Why don't you first try to receive WLAN frames from a normal WLAN card? Then, try replacing the transmitter with an SDR.

YamanDaif, 2021-11-13:
I'm trying to receive its-g5 from Cohda box. 
the sample rate was 20M as it should be for the its-g5 but the results are the same and the .pcap is still empty. I thought that if I change the source/sink in the transceiver file it should work but it seems that this is not the case.

bastibl, 2021-11-14:
ITS-G5 is based on 11p, which uses 10MHz channel, so you should probably set the sample rate to 10MHz.

YamanDaif, 2021-11-15:
Hey!

After getting :
```
RuntimeError: gr::tuntap_pdu::make: tun_alloc failed (are you running as root?)
```
I disabled the the  Tuntap and Ethernet encapsulation and added a message strobe as seen in the screenshot below. It works fine for receiving but not for sending though. Can you find anything wrong in the flowgraph that might be the reason why bladeRF is not sending the message strobe? 
I'm sending its_g5 from BladeRF and receiving on HackRF.
[image]

bastibl, 2021-11-15:
The TUN/TAP interface requires setup as demonstrated in the apps/nic.sh script. There is also a dedicated TX flowgraph that you could try with the osmocom sink. But actually it looks good to me. Check if the other device can receive frames in the first place (e.g. from another Cohda), try different gains (I don't know the valid gain distributions of the BladeRF 2.), check antenna ports etc.

YamanDaif, 2021-11-16:
It works fine now! it was an antenna port problem. Thank you for your fast support!

Yaman

bastibl, 2021-11-16:
Thanks for the feedback. Happy it works.

closed 2021-11-16 by bastibl