bastibl/gr-ieee802-11 #318: WiFi Mapper Block Issue

Mlaws138, 2021-11-23:
I'm having the same issue as #241. The WiFi Mapper block and WiFi frame equalizer blocks are red due to missing definitions of ieee802_11.BPSK_1_2, and chan_est. I installed gnuradio and gr-osmosdr using the command 'sudo apt-get install gnuradio gr-osmosdr' after having trouble getting a source install of gnuradio to work with gr-osmosdr. I'm on Ubuntu 20.04, Python 3.8. I've tried purging gnuradio several times, but am still having the same issues. Thanks in advance for any help.

bastibl, 2021-11-23:
The thread you cited also has a solution. So: Do you have swig installed? Did cmake find swig?

Mlaws138, 2021-11-23:
Sorry, it's a yes to both questions.

bastibl, 2021-11-23:
Can you please make sure that there are no left over libraries and python modules from earlier attempts. Is the correct Python module and library in your path (LD_LIBRARY_PATH and PYTHONPATH)? Did you run `sudo ldconfig`.
You can also try import the module in an interactive Python session (`import ieee802_11` and `dir(ieee802_11)`).
If all of this works, please post the output of the dir command and the cmake output.

closed 2022-04-11 by bastibl