bastibl/gr-ieee802-11 #319: HOW to get CSI

xushanzhizhang, 2021-11-24:
Hi Bastian,

I wanted to know if I can get Channel State Information from wifi_rx.
How to process the obtained file (. dat or. pcap) with MATLAB
I look forward to your reply！！

bastibl, 2021-11-24:
The example flowgraph of recent versions show how CSI can be plotted. The same data can be piped into a file sink and read w/ Matlab. Its a binary file.

xushanzhizhang, 2021-11-24:
I run wifi_ rx.grc(maint_3.7), but there is no response and output.

Is there anything you need to set and pay attention to

bastibl, 2021-11-24:
It was not back-ported to 3.7. You have to update or back-port yourself.

xushanzhizhang, 2021-11-24:
You mean I need maint_3.8 or higher to get CSI?

bastibl, 2021-11-24:
yes.

xushanzhizhang, 2021-11-24:
Thank you very much for your reply. I plan to install maint_3.8 to get CSI

closed 2021-11-24 by bastibl