bastibl/gr-ieee802-11 #321: endless errors

pierregalant, 2021-11-25:
Hello, I would like to use your IEEE transreceiver. Can you tell me which linux version and gnu radio version is the best for your transreceiver. I have tried installing gr-foo with gnuradio 3.8 under kali linux but I am having errors and errors. same for the instant gnuradio. Why you did not install gr-foo and gr-ieee on instant gnuradio ? correct me if I am wrong.

pierregalant, 2021-11-25:
CMake Error at /usr/lib/cmake/pybind11/pybind11Tools.cmake:166 (add_library):
  Target "foo_python" links to target "Log4Cpp::log4cpp" but the target was                                                                                                                                                                
  not found.  Perhaps a find_package() call is missing for an IMPORTED                                                                                                                                                                     
  target, or an ALIAS target is missing?                                                                                                                                                                                                   
Call Stack (most recent call first):                                                                                                                                                                                                       
  /usr/local/lib/cmake/gnuradio/GrPybind.cmake:261 (pybind11_add_module)                                                                                                                                                                   
  python/bindings/CMakeLists.txt:46 (GR_PYBIND_MAKE_OOT)                                                                                                                                                                                   
                                                                                                                                                                                                                                           
                                                                                                                                                                                                                                           
CMake Error at /usr/lib/cmake/pybind11/pybind11Tools.cmake:166 (add_library):
  Target "foo_python" links to target "Boost::filesystem" but the target was                                                                                                                                                               
  not found.  Perhaps a find_package() call is missing for an IMPORTED                                                                                                                                                                     
  target, or an ALIAS target is missing?                                                                                                                                                                                                   
Call Stack (most recent call first):                                                                                                                                                                                                       
  /usr/local/lib/cmake/gnuradio/GrPybind.cmake:261 (pybind11_add_module)                                                                                                                                                                   
  python/bindings/CMakeLists.txt:46 (GR_PYBIND_MAKE_OOT)                                                                                                                                                                                   
                                                                                                                                                                                                                                           
                                                                                                                                                                                                                                           
CMake Error at /usr/lib/cmake/pybind11/pybind11Tools.cmake:166 (add_library):
  Target "foo_python" links to target "Log4Cpp::log4cpp" but the target was                                                                                                                                                                
  not found.  Perhaps a find_package() call is missing for an IMPORTED                                                                                                                                                                     
  target, or an ALIAS target is missing?                                                                                                                                                                                                   
Call Stack (most recent call first):                                                                                                                                                                                                       
  /usr/local/lib/cmake/gnuradio/GrPybind.cmake:261 (pybind11_add_module)                                                                                                                                                                   
  python/bindings/CMakeLists.txt:46 (GR_PYBIND_MAKE_OOT)                                                                                                                                                                                   
[... cut ...]
  Target "gnuradio-foo" links to target "Log4Cpp::log4cpp" but the target was                                                                                                                                                              
  not found.  Perhaps a find_package() call is missing for an IMPORTED                                                                                                                                                                     
  target, or an ALIAS target is missing?

pierregalant, 2021-11-25:
Hello, I have this error under Kali linux with GNU Radio 3.8.2.0 (Python 3.9.1) when I am trying to install gr-foo.

pierregalant, 2021-11-25:
I have tried to change branch to main-3.8 and I have the following error:

CMake Error: File /root/gr-foo/docs/doxygen/Doxyfile.swig_doc.in does not exist.
CMake Error at /usr/local/lib/cmake/gnuradio/GrSwig.cmake:58 (configure_file):
  configure_file Problem configuring file                                                                                                                                                                                                  
Call Stack (most recent call first):                                                                                                                                                                                                       
  /usr/local/lib/cmake/gnuradio/GrSwig.cmake:112 (GR_SWIG_MAKE_DOCS)                                                                                                                                                                       
  swig/CMakeLists.txt:50 (GR_SWIG_MAKE)

bastibl, 2021-11-25:
In the first case you used GNU Radio 3.8 with the 3.9 version of this module. After changing the branch to maint-3.8 delete the build directory and start from scratch.

pierregalant, 2021-11-25:
Thanks for you prompt answer but still struggling :(

CMake Error at /usr/local/lib/cmake/gnuradio/UseSWIG.cmake:418 (add_custom_command):
  Error evaluating generator expression:                                                                          
                                                                                                                  
    $<TARGET_PROPERTY:gnuradio::runtime_swig,INTERFACE_INCLUDE_DIRECTORIES>                                       
                                                                                                                  
  Target "gnuradio::runtime_swig" not found.                                                                      
Call Stack (most recent call first):                                                                              
  /usr/local/lib/cmake/gnuradio/UseSWIG.cmake:559 (SWIG_ADD_SOURCE_TO_MODULE)                                     
  /usr/local/lib/cmake/gnuradio/GrSwig.cmake:133 (swig_add_library)                                               
  swig/CMakeLists.txt:50 (GR_SWIG_MAKE)                                                                           
                                                                                                                  
                                                                                                                  
CMake Error at /usr/local/lib/cmake/gnuradio/UseSWIG.cmake:418 (add_custom_command):
  Error evaluating generator expression:                                                                          
                                                                                                                  
    $<TARGET_PROPERTY:gnuradio::runtime_swig,INTERFACE_INCLUDE_DIRECTORIES>                                       
                                                                                                                  
  Target "gnuradio::runtime_swig" not found.                                                                      
Call Stack (most recent call first):                                                                              
  /usr/local/lib/cmake/gnuradio/UseSWIG.cmake:559 (SWIG_ADD_SOURCE_TO_MODULE)                                     
  /usr/local/lib/cmake/gnuradio/GrSwig.cmake:133 (swig_add_library)                                               
  swig/CMakeLists.txt:50 (GR_SWIG_MAKE)                                                                           
                                                                                                                  
                                                                                                                  
CMake Error at /usr/local/lib/cmake/gnuradio/UseSWIG.cmake:418 (add_custom_command):
  Error evaluating generator expression:                                                                          
                                                                                                                  
    $<TARGET_PROPERTY:gnuradio::runtime_swig,INTERFACE_INCLUDE_DIRECTORIES>                                       
                                                                                                                  
  Target "gnuradio::runtime_swig" not found.                                                                      
Call Stack (most recent call first):                                                                              
  /usr/local/lib/cmake/gnuradio/UseSWIG.cmake:559 (SWIG_ADD_SOURCE_TO_MODULE)                                     
  /usr/local/lib/cmake/gnuradio/GrSwig.cmake:133 (swig_add_library)                                               
  swig/CMakeLists.txt:50 (GR_SWIG_MAKE)                                                                           
                                                                                                                  
                                                                                                                  
CMake Error at /usr/local/lib/cmake/gnuradio/UseSWIG.cmake:418 (add_custom_command):
  Error evaluating generator expression:                                                                          
                                                                                                                  
    $<TARGET_PROPERTY:gnuradio::runtime_swig,INTERFACE_INCLUDE_DIRECTORIES>                                       
                                                                                                                  
  Target "gnuradio::runtime_swig" not found.                                                                      
Call Stack (most recent call first):                                                                              
  /usr/local/lib/cmake/gnuradio/UseSWIG.cmake:559 (SWIG_ADD_SOURCE_TO_MODULE)                                     
  /usr/local/lib/cmake/gnuradio/GrSwig.cmake:133 (swig_add_library)                                               
  swig/CMakeLists.txt:50 (GR_SWIG_MAKE)                                                                           
                                                                                                                  
                                                                                                                  
[... cut ...]
    $<TARGET_PROPERTY:gnuradio::runtime_swig,INTERFACE_INCLUDE_DIRECTORIES>                                       
                                                                                                                  
  Target "gnuradio::runtime_swig" not found.                                                                      
Call Stack (most recent call first):                                                                              
  swig/CMakeLists.txt:50 (GR_SWIG_MAKE)

bastibl, 2021-11-26:
It looks like you compiled GNU Radio yourself but didn't have swig installed while doing it. Another explanation is that you tried several installation options and versions, and ended up with left-over artifacts from earlier attempts. Make sure to get rid of installed artifacts or installations from the package manager when you compile yourself.

pierregalant, 2021-11-28:
Nope. I am trying to install gr-foo with a binary version of gnuradio...I tried removing the artefacts, but I am still having errors.

bastibl, 2021-11-29:
You open tons of issues in my repositories without a clear description of what you are doing and issues that only arise to you. You say you use a binary version of GNU Radio but paths point to `/usr/local`. This is not where distribution packages are installed, which I think you mean when referring to binary packages. Please restart from a clean environment. If you still have errors, please open one issue with the exact steps and versions of packages you installed and how.

closed 2021-11-29 by bastibl