bastibl/gr-ieee802-11 #322: Compiling errors with instant gnuradio

pierregalant, 2021-11-25:
Hello, I am also trying to install gr-foo maint-3.8 on an instant gnuradio with gnuradio 3.8. I have already install libpcl-dev

However, I am having the followings errors: [  7%] Building CXX object lib/CMakeFiles/gnuradio-foo.dir/burst_tagger_impl.cc.o
[ 15%] Building CXX object lib/CMakeFiles/gnuradio-foo.dir/channel_model_impl.cc.o
[ 23%] Building CXX object lib/CMakeFiles/gnuradio-foo.dir/packet_dropper_impl.cc.o
[ 30%] Building CXX object lib/CMakeFiles/gnuradio-foo.dir/packet_pad2.cc.o
[ 38%] Building CXX object lib/CMakeFiles/gnuradio-foo.dir/packet_pad_impl.cc.o
[ 46%] Building CXX object lib/CMakeFiles/gnuradio-foo.dir/periodic_msg_source_impl.cc.o
[ 53%] Building CXX object lib/CMakeFiles/gnuradio-foo.dir/random_periodic_msg_source_impl.cc.o
[ 61%] Building CXX object lib/CMakeFiles/gnuradio-foo.dir/rtt_measure_impl.cc.o
[ 69%] Building CXX object lib/CMakeFiles/gnuradio-foo.dir/wireshark_connector_impl.cc.o
[ 76%] Building CXX object lib/CMakeFiles/gnuradio-foo.dir/pad_tagged_stream_impl.cc.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgnuradio-foo.so.7557bf48'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:206: lib/CMakeFiles/gnuradio-foo.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

bastibl, 2021-11-26:
Where did you download the VM?

pierregalant, 2021-11-26:
From https://github.com/bastibl/instant-gnuradio

bastibl, 2021-11-29:
I don't know what you installed, but the library seems to miss liborc-0.4. Installing it `sudo apt install liborc-0.4-dev` fixes it.

closed 2021-11-29 by bastibl