bastibl/gr-ieee802-11 #323: ieee802_11 missing blocks moving average

pierregalant, 2021-11-28:
Hello, I have successfully install gnuradio on a virtual machine with ubuntu. However, when I am trying to open a grc. I have an error with some block with error message: missing: ieee802_11 moving average. 

I have tried to export my python path and when I type python then import ieee802_11, I do not have an error message. How to fix this error on gnuradio
[image]
 ?

bastibl, 2021-11-29:
You opened the flowgraph without installing the module (or you opened a flowgraph that doesn't work with the version you have installed).

closed 2021-11-29 by bastibl

pierregalant, 2021-11-29:
Hello for this flowgraph, I am on ubuntu latest version LTS in a VM. I have installed gnuradio 3.8.1.0 (python 3.8.10) binary version. The flowgraph is a traditional wifiTX.grc. Is it compatible ?

Moreover, do you know where should I set the LD_LIBRARY_PATH on ubuntu. Maybe this is the problem, because, I have found that it was the case for some users.

bastibl, 2021-11-29:
The above problem is that GRC cannot find the block definition. If you didn't set the library path, there will anyway be follow-up problems. However, since none of these problems are specific to this module, I'd ask you to write to the GNU Radio mailing list or join the Matrix chat.