bastibl/gr-ieee802-11 #326: wifi_phy_hier.grc Error!

SoulH-qqq, 2021-12-16:
Hi,
thank you for your amazing software. I have compiled gr-foo 3.8 and gr-802.11 3.8. My system is Ubuntu 18.04. My gnuradio version is as follows:
[image]
I also installed swig！But when I open wifi_phy_hier.grc, the following problems still occur!
[image]
I don't know what to do, can you help me! !

Best,
Hao Sun

bastibl, 2021-12-16:
Please see the readme. You have to set the block path and configure environment variables.

SoulH-qqq, 2021-12-17:
I made the following changes according to the readme:
[image]
When I tried to open the /usr/local/share/gnuradio/grc/blocks directory, I found the following files:
[image]
Only files starting with “foo”. I found that these files are the same as those in the gr-foo-maint-3.8/grc directory. So I tried to copy the files in the /gr-ieee802-11-maint-3.8/grc directory to the /usr/local/share/gnuradio/grc/blocks/ directory, the results are as follows:
[image]
But when I open wifi_phy_hier.grc the error is not resolved.
[image]

SoulH-qqq, 2021-12-17:
The ieee802.11 module can be searched in grc:
[image]

bastibl, 2021-12-17:
You have to set the environment variables (PYTHONPATH, LD_LIBRARY_PATH) as described in the readme.

SoulH-qqq, 2021-12-18:
Sorry, I did not see how to set the environment variable in the readme, can you tell me how to set it, or the location of the setting instructions.

bastibl, 2021-12-20:
It's like this: https://www.wime-project.net/installation/#adapt-environment-variables but with the paths that you used. So don't just blindly copy paste. Also, this is the same for each OOT. So you can Google, check the GNU Radio wiki, etc.

SoulH-qqq, 2021-12-20:
I have reconfigured the environment variables according to https://wiki.gnuradio.org/index.php/InstallingGR.
[image]
When I open the Python interaction, import 802_11 still fails. Opening wifi_phy_hier.grc is still the original error.

bastibl, 2021-12-20:
So the obvious question: Is the module actually installed under `/usr/lib/python3/dist-packages`? Is the lib under `/usr/lib`. I guess not... That's what I meant with don't just blindly copy-paste commands....

SoulH-qqq, 2021-12-20:
[image]
This shows that gnuradio is installed in the /usr directory.
[image]
This indicates the location of the Python library.
So I set environment variables in the .bashrc file:
export PYTHONPATH=/usr/lib/python3/dist-packages
export LD_LIBRARY_PATH=/usr/lib

bastibl, 2021-12-20:
GNU Radio already works. The question where you installed the OOT, i.e., this module. Please write to the GNU Radio mailing list, if you have any other question. I cannot provide this level of support at the moment.

closed 2021-12-20 by bastibl