bastibl/gr-ieee802-11 #328: PER for different modulation scheme.

fancyshirt, 2022-02-24:
I'm using the module with USRP N310 on 2.4 GHz band with 20 MHz bandwidth.
Both BPSK and QPSK has high packet delivery rate, up to 91-95%.
However, when I use 16QAM or 64QAM, the packet delivery rate drop to 54% and 1.3%, respectively.
I have tried different tx and rx gain pairs (normalized), but the best I can get are as mentioned.
I wonder if there is any setting I should change?

bastibl, 2022-02-24:
I'm afraid you'd have to figure out the underlying issue first. Overruns because CPU cannot keep up, issues streaming samples, SNR, interference, etc. etc. You have to figure out what goes wrong.

fancyshirt, 2022-02-24:
Thanks for your reply. I will go check if my hardware is sufficient for the experiment.

closed 2022-03-09 by fancyshirt