bastibl/gr-ieee802-11 #330: How to format an ACK frame

fancyshirt, 2022-03-09:
I'm trying to use scapy to construct my frame.
I can use the library to create beacon, auth, and many other frame and be received by gr-ieee802-11 without a problem.
However, since there is no data in ACK frame, the checksum always failed to pass the condition. (in ./lib/decode_mac.cc)
How should I construct the ACK, so the checksum can pass the condition?

Many thanks

bastibl, 2022-03-09:
So you get "checksum wrong -- dropping"? I've seen decoded ACK frames over the air, so I have no idea what should be the issue. Do you send data over-the-air? If so, can you receive the ACK with another WLAN card, i.e., are you sure that it is an issue of this module?

fancyshirt, 2022-03-09:
Thanks for your reply.

After digging more of my approach, I decided to create my own ACK frame.
Previously the ACK frame created by scapy does not have any attribute, such as type, subtype and RA.
Right now, I'm following this page: https://howiwifi.com/2020/07/13/802-11-frame-types-and-formats/
I manually add the type, subtype and RA into the frame I create, and it passed the decode_mac.cc.
However, I encounter another problem in parse_mac.cc.

In parse_mac.cc, it first check if the frame length is less than 20 or not.
If it is less than 20, it says 'frame too short to parse (<20)'
But, according to the page I cited above, the ACK always has only 10 bytes, excluding the FCS.
Therefore, it always fails to pass the condition.

I wonder if the condition expression in parse_mac.cc should first check if it has type of control and subtype of ACK, and then check the length, so the ACK can successfully be parsed.

I comment out the condition expression in parse_mac.cc, line 69 to 72, and now everything works fine.
I'm not sure if my approach is in the right direction or I misunderstand something.

bastibl, 2022-03-10:
OK, so my understanding is that the initial problem was not related to this module.

The Parse MAC block is just for quick debugging. It's not a full-fledged parser for WLAN frames. I'd recommend to just pipe the data in a Wireshark sink and look at it there.

If you want to improve/extend the Parse MAC block, feel free to open a PR. But it will probably never be a complete parser for all WLAN frames.

fancyshirt, 2022-03-10:
Thanks for your reply.
I have created my own module for my reseach.
Thanks again for clarify things for me.
I'll close this issue.

closed 2022-03-10 by fancyshirt