bastibl/gr-ieee802-11 #331: compile issues on Mac M1

MAVProxyUser, 2022-03-20:
Have you folks been able to get this working on OSX m1 Macs? 

MacBook-Air:gr-ieee802-11 kfinisterre$ make
[  4%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/chunks_to_symbols_impl.cc.o
/Users/user/Desktop/gr-ieee802-11/lib/chunks_to_symbols_impl.cc:27:12: error: no viable conversion from returned value of type 'std::shared_ptr<chunks_to_symbols_impl>' to function return type 'chunks_to_symbols::sptr' (aka 'shared_ptr<gr::ieee802_11::chunks_to_symbols>')
    return gnuradio::get_initial_sptr(new chunks_to_symbols_impl());
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/homebrew/include/boost/smart_ptr/shared_ptr.hpp:352:21: note: candidate constructor not viable: no known conversion from 'std::shared_ptr<chunks_to_symbols_impl>' to 'boost::detail::sp_nullptr_t' (aka 'nullptr_t') for 1st argument
    BOOST_CONSTEXPR shared_ptr( boost::detail::sp_nullptr_t ) BOOST_SP_NOEXCEPT : px( 0 ), pn()
                    ^
/opt/homebrew/include/boost/smart_ptr/shared_ptr.hpp:458:5: note: candidate constructor not viable: no known conversion from 'std::shared_ptr<chunks_to_symbols_impl>' to 'const boost::shared_ptr<gr::ieee802_11::chunks_to_symbols> &' for 1st argument
    shared_ptr( shared_ptr const & r ) BOOST_SP_NOEXCEPT : px( r.px ), pn( r.pn )
    ^
/opt/homebrew/include/boost/smart_ptr/shared_ptr.hpp:670:5: note: candidate constructor not viable: no known conversion from 'std::shared_ptr<chunks_to_symbols_impl>' to 'boost::shared_ptr<gr::ieee802_11::chunks_to_symbols> &&' for 1st argument
    shared_ptr( shared_ptr && r ) BOOST_SP_NOEXCEPT : px( r.px ), pn()
    ^
/opt/homebrew/include/boost/smart_ptr/shared_ptr.hpp:486:5: note: candidate template ignored: could not match 'boost::shared_ptr' against 'std::shared_ptr'
    shared_ptr( shared_ptr<Y> const & r, typename boost::detail::sp_enable_if_convertible<Y,T>::type = boost::detail::sp_empty() )
    ^
/opt/homebrew/include/boost/smart_ptr/shared_ptr.hpp:520:5: note: candidate template ignored: could not match 'auto_ptr' against 'shared_ptr'
    shared_ptr( std::auto_ptr<Y> && r ): px(r.get()), pn()
    ^
/opt/homebrew/include/boost/smart_ptr/shared_ptr.hpp:552:5: note: candidate template ignored: could not match 'unique_ptr' against 'shared_ptr'
    shared_ptr( std::unique_ptr< Y, D > && r ): px( r.get() ), pn()
    ^
/opt/homebrew/include/boost/smart_ptr/shared_ptr.hpp:568:5: note: candidate template ignored: could not match 'unique_ptr' against 'shared_ptr'
    shared_ptr( boost::movelib::unique_ptr< Y, D > r ): px( r.get() ), pn()
    ^
/opt/homebrew/include/boost/smart_ptr/shared_ptr.hpp:679:5: note: candidate template ignored: could not match 'boost::shared_ptr' against 'std::shared_ptr'
    shared_ptr( shared_ptr<Y> && r, typename boost::detail::sp_enable_if_convertible<Y,T>::type = boost::detail::sp_empty() )
    ^
/opt/homebrew/include/boost/smart_ptr/shared_ptr.hpp:371:14: note: explicit constructor is not a candidate
    explicit shared_ptr( Y * p ): px( p ), pn() // Y must be complete
             ^
/opt/homebrew/include/boost/smart_ptr/shared_ptr.hpp:465:14: note: explicit constructor is not a candidate
    explicit shared_ptr( weak_ptr<Y> const & r ): pn( r.pn ) // may throw
             ^
/opt/homebrew/include/boost/smart_ptr/shared_ptr.hpp:507:14: note: explicit constructor is not a candidate
    explicit shared_ptr( std::auto_ptr<Y> & r ): px(r.get()), pn()
             ^
1 error generated.
make[2]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/chunks_to_symbols_impl.cc.o] Error 1
make[1]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/all] Error 2
make: *** [all] Error 2

bastibl, 2022-03-21:
Did you make sure that you use the correct version of the module, i.e., `maint-3.8` branch with GNU Radio 3.8 etc?

MAVProxyUser, 2022-03-21:
Brew installs GnuRadio 3.9.3  & Boost 1.78. 
https://formulae.brew.sh/formula/boost
https://formulae.brew.sh/formula/gnuradio

I'll give it a shot with 3.8 and check back.

MAVProxyUser, 2022-03-25:
maint-3.9 + brew seemed to be ok. Still have a few pathing issues to sort out, but compile is fine now!

closed 2022-03-25 by MAVProxyUser