bastibl/gr-ieee802-11 #332: Stuck in running examples/wifi_transceiver

gcc17, 2022-04-06:
Hi, the running of `examples/wifi_transceiver` is stuck in my testing. It keeps printing 'O' on the terminal. 
branch: maint-3.7
system: ubuntu 16.04
device: USRP B210

Screenshot for running `wifi_transceiver`:
[image]

The running of `examples/wifi_loopback` looks good:
[image]

I cannot figure out where it is that the testing gets stuck. When running `wifi_transceiver`, the memory usage shown by `top` is less than 10%, so the memory should not be the bottleneck. I would appreciate it a lot if you can offer some advice for debugging.

gcc17, 2022-04-06:
I repeated the same installation process on another computer, and the testing of `wifi_transceiver` worked well. 
[image]

Both computers are Ubuntu 16.04 and their GNU Radio versions are the same... Are there any other factors can cause the difference?

bastibl, 2022-04-11:
Memory doesn't matter too much, it's the CPU that's important. If the CPU cannot keep up with the samples, you experience overruns, and streaming is restarted all the time. Also running `volk_profile` could make a difference. Make sure that it was run on the first PC.

gcc17, 2022-04-13:
Thanks for your reply! I have runned `volk_profile` on both PCs. It seems that the generated files have a few differences. Do these differences matter? Actually the first PC has a newer processor than the second one, while the first one has 8 GB memory and the second has 16 GB. 
[image]

bastibl, 2022-04-13:
Memory is not the problem. The fact that different implementations are preferable on different CPUs is exactly why you have to run `volk_profile`. So this is also as expected. One PC is just not fast enough, it seems.

closed 2022-08-02 by bastibl