bastibl/gr-ieee802-11 #335: Issue with File Sink

obsidian114, 2022-04-18:
While running wifi_rx.grc,  I am getting output while File Sink is disabled, Once I enable Wireshark connector and File Sink I'm getting the following error:

gr::log :WARN: flowgraph - Could not find port in in:
gr::log :WARN: flowgraph -   in
gr::log :WARN: flowgraph -   system
Traceback (most recent call last):
  File "/home/sj/gr-ieee802-11/examples/wifi_rx.py", line 423, in <module>
    main()
  File "/home/sj/gr-ieee802-11/examples/wifi_rx.py", line 401, in main
    tb = top_block_cls()
  File "/home/sj/gr-ieee802-11/examples/wifi_rx.py", line 298, in __init__
    self.msg_connect((self.ieee802_11_decode_mac_0, 'out'), (self.foo_wireshark_connector_0, 'in'))
  File "/usr/local/lib/python3.8/dist-packages/gnuradio/gr/hier_block2.py", line 51, in wrapped
    func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
  File "/usr/local/lib/python3.8/dist-packages/gnuradio/gr/hier_block2.py", line 127, in msg_connect
    self.primitive_msg_connect(*args)
ValueError: invalid msg port in connect() or disconnect()

bastibl, 2022-04-21:
This happened to me, if I used flowgraphs/hier blocks (e.g., PHY Hier) from different GNU Radio versions.

obsidian114, 2022-04-25:
Thanks for the reply. Any solutions for this?

bastibl, 2022-04-25:
Make sure you have only one version of  GNU Radio installed and no artifacts from other installation attempts. Install the version of this module that matches the version of GNU Radio. Open the flowgraph for the corresponding version.

obsidian114, 2022-04-25:
Thanks Bastian

closed 2022-04-25 by obsidian114