bastibl/gr-ieee802-11 #337: Multi antenna reception of WiFi signal

Leoplt, 2022-04-28:
Hi,bastibl
    Thank you for your work.
    I have used USRP N210 to obtain the CSI information of the received signal successfully.
    Now, I want to get the CSI information of multiple received signals at the same time.
    What should I do?

bastibl, 2022-04-28:
Create a USRP with two (or more) outputs for the different antennas https://kb.ettus.com/Synchronization_and_MIMO_Capability_with_USRP_Devices and duplicate the whole GNU Radio DSP chain until CSI extraction. If you don't have a good PC, this will likely lead to performance issues. Also without further adaption, you might have correlate data manually. Since both receivers operate independently, one might detect a frame, while the other misses the frame. So the n-th CSI value of one receive chain might not correspond to the n-th CSI of the other receive chain.

Leoplt, 2022-05-09:
Thank you for your reply.
The device I use is USRP X310, which has two receiving antennas at the same time.
I want to know how to distinguish the information collected by the two antennas?
Are they stored alternately in .dat file?
In addition, can you provide the corresponding matlab program to process the data

bastibl, 2022-05-09:
There is no dat file. There is no matlab. Also, an SDR with more than one receive streams doesn't just turn this in a MIMO receiver. I think you should familiarize yourself a bit more with the project and the topic.

Leoplt, 2022-05-12:
Two antennas collect CSI at the same time.
Are they simultaneous? Alternating? Random? Competition?

bastibl, 2022-05-12:
http://www.catb.org/~esr/faqs/smart-questions.html

closed 2022-05-12 by bastibl