bastibl/gr-ieee802-11 #338: Transmission power range of USRP N210

xushanzhizhang, 2022-05-02:
Hi,Bastibl:
    I'm bothering you
   When using USRP N210 to transmit signals, the transmission power can be changed by adjusting gain
   What is the range of transmission power? What is the corresponding relationship with gain?

bastibl, 2022-05-02:
This is a question for Ettus and not really related to this project. Anyways, you could have a look at their [knowledge base](https://kb.ettus.com/Knowledge_Base). In short, it depends on your daughter board.  You can check the gain range with `uhd_usrp_probe`. Usually, USRPs are not calibrated, i.e., you cannot easily map gain to output power. But AFAIK they introduced an API that might enable this (at least give an estimate of the output power). You have to check the wiki and API docs.

closed 2022-05-02 by bastibl

xushanzhizhang, 2022-05-06:
Is this item applicable to USRP X310

I want to use USRP X310 to receive WiFi signals

thanks

 
Closed #338.

bastibl, 2022-05-06:
Yes.