bastibl/gr-ieee802-11 #340: ImportError:/usr/local/lib/libgnuradio-foo.so.1.1.0git:undefined symbol

queenandking, 2022-05-05:
[image]
Hi, I meet this problem after I create my block include gr-foo.
I don't understand what's meaning.
Could you help me solve it?
Thank you so much!

queenandking, 2022-05-05:
Hi, bastibl,

Thank you，I have solve that problems.
But I have another problem about the function 
 message_port_pub(pmt::pmt_t port_id, pmt::pmt_t msg)
This is the orignal function.
I see that mac.cc in gr-ieee802-15-4 have used this function by message_port_pub(pmt::mp("pdu out"),pmt::cons(pmt::PMT_NIL,pmt::make_blob(d_msg,d_msg_len))).
I use the same function like yours, it not worked and show errors without objective pmt::cons(pmt::PMT_NIL,pmt::make_blob(d_msg,d_msg_len).
Could you tell me how to solve this error?
Thank you!

bastibl, 2022-05-06:
Sorry, but I have no idea what you're talking about. These are, however, just C++/GNU Radio problems. You could write to the GNU Radio mailing list or their Matrix chat. Also it might make sense share a link to the repo and the particular part of the code that you're talking about. Overall, it sounds like you are just not very experienced with C++.