bastibl/gr-ieee802-11 #344: Modify Transmitter Protocal, specifically I want to decrease the symbol rate of the long and short training sequnce

wenjinzhang, 2022-06-02:
Hi,

Thank you for sharing the amazing project. I am trying to use a third device to detect the short training sequence and issue the attack signal to interfere with the long training sequence. The challenge is that the time frame is too short because short and long training symbols are sent in 8us+8us, while the latency between USRP and PC can reach 1ms. I think I have to implement this on FPGA. Right now, I am thinking about whether I can increase the time frame by decreasing the symbol rate or not. For example, I want to make short training sequences sent in 8ms rather than 8us. Would you please give me some guidance on how to control the symbol rate? Thanks in advance!

bastibl, 2022-06-03:
You'd have to adapt the sync words of the Carrier Allocator block. Alternatively, you could use a normal WLAN dongle https://dl.acm.org/doi/abs/10.1145/2664243.2664260

wenjinzhang, 2022-07-27:
Hi

Thank you for your information. By modifying the long training sequence of sync words, I can change the CSI. Specifically, when I double the power of the long training sequence (X2), the CSI amplitude of all 52 subcarriers is also doubled (X2). I want to further modify the CSI amplitude of some subcarriers, not all subcarriers. How can I modify the sync words to change the CSI amplitude of one specific subcarrier?  How is the sync word transmitted through 52 subcarriers?

Thank you very much for your help.

Best regards,
Wenjin

bastibl, 2022-07-28:
Calculating the sync words from the frequency domain representation is a bit involved, given the unusual structure of the long preamble. You can see how they are calculated here: https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/utils/sync_words.py
You can adapt the power level and then put them in the carrier allocator.

wenjinzhang, 2022-07-28:
Hi,

Thank you for your guidance. If I understand correctly, I need to transform the long preamble in ```sync_word.py``` from time domain to the frequency domain. Then, I can adapt the power level of each frequency band. Next, I can transform the frequency representation back to the time domain.

Best regards,
Wenjin

bastibl, 2022-07-29:
In the script, the short and long preamble are set in *frequency* domain. So you would change the subcarrier as needed, regenerate the preamble, and put it in the carrier allocator.

wenjinzhang, 2022-07-29:
It makes sense. Thank you very much.

wenjinzhang, 2022-08-02:
Is there a way to update the sync word in real-time? Specifically, I generate a list of sync words, ["sync words1", "sync words2", "sync words3", "sync words4"] ahead of time. When Tx starts to transmit, Tx will use the first sync word, "sync words1" of the list to generate the first WiFi packet and the second sync word, "sync words2" to generate the second WiFi packet. Below is my current thought. 

I added a new parameter called 'str_sync_words' on "WiFi PHY Hier". Then, I can configure the "sync word" in wifi_tx.grc as shown below, and the generated python script "wifi_tx.py" also have this configurable parameter as shown in the screenshot.
[image] 
[image]

The sync word is hard coded on the generated python script. I plan to change it in real-time. However, I found the loop of the Tx side is controlled by a "message strobe" block with a predefined interval. I don't know how to update the sync word in real-time. I cannot find the loop inside the generated python script. Would you please advise me? Thanks.

Best regards,
Wenjin

bastibl, 2022-08-04:
The sync words are put in the Carrier Allocator block. This is a block from upstream GNU Radio; not this repository. I'd recommend to either directly modify the block in your GNU Radio fork or copy the block to an out-of-tree module and extended it with the functionality that you need. You can check other blocks, how callbacks (for live reconfiguration) work. Since this is a general GNU Radio problem, you can check the wiki or ask on the mailing list.

wenjinzhang, 2022-08-04:
Thank you very much for your guidance. I will check other blocks and explore how callbacks work. I will check the wiki and ask on the mailing list.

closed 2022-08-04 by wenjinzhang