bastibl/gr-ieee802-11 #347: Signal transmission time and interval

lrisd, 2022-06-25:
Hello bastibl:
Thank you for your work!
I successfully implemented example wifi_tx.grc on USRP
I set the encoding method as BPSK 1/2, the interval is 10, and the packet length is 300.
What is the estimated transmission time and transmission interval of WiFi packets?

bastibl, 2022-06-25:
It depends on your PC. There are no timing guarantees and the transmitter is not optimized for maximum throughput.

lrisd, 2022-07-05:
Thank you for your reply!
Adjust 'tx_gain' only without changing the hardware device and parameters 'encoding','interval','pdu_length', etc.
Will its transmission rate change?

bastibl, 2022-07-05:
The interval determines the rate. The other parameters have no impact, if the transmitter is not running at full CPU.

closed 2022-08-02 by bastibl