bastibl/gr-ieee802-11 #348: Packet Pad2 block's function

romm32, 2022-06-27:
Hi!
I have been using this module for the past couple of weeks to work on a Digital Communications' project from my university. 
I am having trouble understanding what exactly it is that the block Packet Pad2 does. I have tried reading its documentation and searching on related work (Performance Assessment of IEEE 802.11p with an Open Source SDR-based Prototype), concluding that it must be adding the short and long preambles in some way. However, I don't understand how it does it (given it does). Is it just adding zeros to the beginning of each packet? In this case, I don't completely get why its default value is 500. Shouldn't it be 160(short preamble)+160(long preamble)=320? It may well be that the default value is not actually meant to be default. In that case, sorry for the inconvenience.
Thank you so much in advance.
Romina

bastibl, 2022-06-27:
Hi, it's open source, so you can just look at the code. It's basically two lines:

https://github.com/bastibl/gr-foo/blob/maint-3.9/lib/packet_pad2.cc#L57-L59

It adds zeros before and after a packet. This is necessary for some SDRs, since they need some time to stabilize the oscillator and, therefore, have transients during the first samples, which might disturb the frame. This can happen for packet-based communication, where streaming restarts for every frame that is sent.

romm32, 2022-06-28:
Thank you for your answer. I understand now

closed 2022-06-29 by bastibl