bastibl/gr-ieee802-11 #349: Just a general question about the blocks

jackdvd, 2022-07-04:
First of all thanks for sharing this. 
Sorry, if this sounds like a stupid Q as I am new to SDR. Why is the block diagram so complicated and doesn't use any direct filter like LPF etc? Is this because the encoding of the WiFi itself is complicated? Is there any document explaining the role of each blocks?
Is there a simple way to achieve this like using blocks like LPF, GFSK demod etc.?

bastibl, 2022-07-05:
There is no simple way to implement a WLAN receiver with just some basic components. Yet, for example, the moving average blocks are a LPF. Or the Sync Long block uses a cross-correlation FIR filter internally. So these components might be in there somewhere, but there's some more complexity to it that requires dedicated/specialized blocks.

closed 2022-07-05 by bastibl

jackdvd, 2022-07-06:
Thanks for the response. Could you please recommend some resource for understanding the internals of WiFi signals. Tried to google it but not coming across any good ones.