bastibl/gr-ieee802-11 #350: Can send packet but can receive well

rom1tes, 2022-07-06:
Hello,

I've been using your project to sniff some V2X messages (Based on 802.11p standard). The reception works well. The next step is to send some messages back, but I can figure out why it does not work. It does not seems to encode correctly. 
For instance this is what I receive (with a bladeRF) from the working device :
[image]
As you can see, the reception works fine and I can save the messages that are sent. 
On the other hand, This is what I receive (with a bladeRF) when I send something with (another bladeRF) :
[image]
I have tried to do volk_profile but I get the same result. As you said in another issue, It might be related to the CPU, with the wifi_tx on, my CPU is not overload at all (5% in total) so is there a way to boost the CPU consumption ?
I have the same problem on 2 computer with different installation.

Thank you very much.

bastibl, 2022-07-06:
The auto-correlation is constantly high, which triggers frame detection all the time. This causes load and makes missing frames much more likely. You can try to increase the distance. Shield the transmitter. Check if there are some dc offset correction settings.

closed 2022-08-02 by bastibl

liiso7, 2024-02-28:
@rom1tes Hello, how do you use bladeRF to sniff messages? Have you tried any other frequencies? Can you provide me with your.GRC flow chart? Thank you very much!