bastibl/gr-ieee802-11 #352: CFO and SFO

AlanerLin, 2022-08-07:
hello bastibl:
I want to collect the data of CFO(Carrier frequency offset) and SFO(sampling frequency offset).
I installed branch of maint-3.7 for my computer.
Can I find data inside /gr-ieee802-11/lib/*.cc ?
Such as variable "d_freq" came from /gr-ieee802-11/lib/frame_equalizer_impl.h.

bastibl, 2022-08-10:
The frequency offset is annotated as metadata for the frame: https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/frame_equalizer_impl.cc#L229-L231. These tags are also forwarded by later stages (https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/decode_mac.cc#L76-L78). You can just add a Message Debug block after the decoder to get the metadata printed. SFO is not estimated separately but assumed to correlate with the CFO.

AlanerLin, 2022-08-11:
Thank you for your answer.

closed 2022-08-11 by AlanerLin