bastibl/gr-ieee802-11 #356: gnuradio/tag_checker.h: No such file or directory

JeanPaulBadjo, 2022-08-30:
For version 3.10 When trying to execute "make" in gr-ieee802-11/build the error below appears: 

[image]

JeanPaulBadjo, 2022-08-30:
I have looked at other issues and I think I found the solution to this. It was to remove the tag_checker portion from the chunks_to_symbols file. 
I have since run into a new problem below:

-- Extracting version information from git describe...
fatal: No names found, cannot describe anything.
-- Using install prefix: /usr/local
-- Building for version: gef5fd6c / 1.1.0git
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.10") 
CMake Error at swig/CMakeLists.txt:34 (include):
  include could not find load file:

    GrSwig

CMake Error at swig/CMakeLists.txt:48 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

--------------

Now I tried to install GrSwig but it seems it's already installed on my computer. I looke again for other possible solutions and in one post you stated that there could be artifacts from previous installs of different versions. So I will wipe my computer and reinstall ubuntu and start from scratch to make sure.

bastibl, 2022-08-31:
You have to use the `maint-3.10` branch of this module with GNU Radio 3.10.

JeanPaulBadjo, 2022-09-02:
Ok I played around with it and managed to fix it. I had downloaded the wrong file. I hadn't noticed that I had to directly download the different versions from a different page.  But now I'm going to use 3.8. Thanks so much! I'm having a separate issue now so Ill open a new "issue".

closed 2022-09-02 by JeanPaulBadjo