bastibl/gr-ieee802-11 #359: Value cannot be evaluated or defined

JeanPaulBadjo, 2022-09-02:
I'm having the errors below when trying to run wifi_phy_hier:

[image]
[image]

I did some googling and I saw a number of responses from running the code through some IDE to see if its picking up the SO, to making sure the blocks and Library's are all communicating but it hasnt worked, and I'm worried that I might be typing in the commands wrong.  Below is the locations to the SO file, the blocks, and the libraries. 

[image]

Please help me

bastibl, 2022-09-04:
This is an installation issue. However, the information and the screenshots to not help much debugging this. Please see the readme and make sure that the various paths are set correctly.

JeanPaulBadjo, 2022-09-08:
Ok, I ran into the same problem, however i went to version 3.10 and it worked perfectly. I have a few other questions but I'll open a new question.

closed 2022-09-08 by JeanPaulBadjo