bastibl/gr-ieee802-11 #361: are you as root?

mbalalzade, 2022-09-14:
hi
i used this repos.(ieee802.11) with bladeRF. but i have 2 problem with this. first is this function:
>>>>>uhd.tune_request(freq, rf_freq = freq - lo_offset, rf_freq_policy=uhd.tune_request.POLICY_MANUAL)) <<<<<<<
that use in usrp _sink and usrp_source  and i dont know how does it work . does it possible to use just freq param that there is in program instead of that or not because i used osmo lib. instead of UHD lib . another problem is when i try to run tranciever.grc it say to me      >>>>>>>>>>are you as root?<<<<<<<<<<<
also i have some questions like we all hear about the max power that wifi send on air usually is 1 watt does it means the RF gain in dB in osmo_sink and osmo_source should be 0d B or this parameter have another meaning.

thanks

bastibl, 2022-09-14:
If your SDR does not support offset tuning, just use the `freq` variable as your frequency. 

The problem with root happen because of the TUN/TAP block. If you don't need it, just disable it. If you need it, you can setup the interface in advance. See the scripts in the apps directory. Then you shouldn't need root privileges for the flowgraph.

mbalalzade, 2022-09-14:
Thanks alot

closed 2022-09-14 by mbalalzade