bastibl/gr-ieee802-11 #362: installation problem of gr-ieee-80211, version 3.9

franckstik, 2022-09-14:
Hi,
when I try to install gr-ieee-80211 on Ubuntu 20.04, with pybombs, with gnu radio version 3.9, I get the following message error :
-------------------
[INFO] Phase 2 complete: All source packages installed.
[INFO] Prefix Python version is: 3.8.10
[INFO] PyBOMBS Version 2.3.5
[INFO] Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gr-ieee-80211
   |
   \- gr-foo
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: gr-foo
warning: pas de commit commun
remote: Enumerating objects: 1051, done.
remote: Counting objects: 100% (169/169), done.
remote: Compressing objects: 100% (81/81), done.
remote: Total 1051 (delta 85), reused 165 (delta 83), pack-reused 882
Réception d'objets: 100% (1051/1051), 336.62 Kio | 1.92 Mio/s, fait.
Résolution des deltas: 100% (642/642), fait.
Depuis https://github.com/bastibl/gr-foo
 * [nouvelle branche]      maint-3.10 -> gr-foo/maint-3.10
 * [nouvelle branche]      maint-3.7  -> gr-foo/maint-3.7
 * [nouvelle branche]      maint-3.8  -> gr-foo/maint-3.8
 * [nouvelle branche]      maint-3.9  -> gr-foo/maint-3.9
 * [nouvelle étiquette]    deprecated_next_branch -> deprecated_next_branch
Clonage dans 'gr-foo'...
Configuring: (100%) [=======================================================================================================================================================================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
-- PyBOMBS installed GNU Radio. Setting CMAKE_INSTALL_PREFIX to /home/tim/prefix-3.9
-- UHD library found.
-- Using GMP.
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.10") 
-- Extracting version information from git describe...
-- Using install prefix: /home/tim/prefix-3.9
-- Building for version: 7557bf48 / 1.1.0git
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.10") 
CMake Error at swig/CMakeLists.txt:36 (include):
  include could not find load file:

    GrSwig

CMake Error at swig/CMakeLists.txt:50 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

-- Configuring incomplete, errors occurred!
See also "/home/tim/prefix-3.9/src/gr-foo/build/CMakeFiles/CMakeOutput.log".
See also "/home/tim/prefix-3.9/src/gr-foo/build/CMakeFiles/CMakeError.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gr-foo:
Configuration failed
[ERROR] Error installing package gr-foo. Aborting.
-------------
Would you have the possibility to correct the problem ?
Thank you

bastibl, 2022-09-14:
It looks like you are trying to compile the 3.8 oder 3.7 version of this module with the 3.9 version of GNU Radio, which fails. I'd just build it manually without PyBombs. Not sure if this is still maintained.

franckstik, 2022-09-19:
Hello Bastian,
when I try to compile the sources I get the following error message :
*******
Clonage dans 'gr-ieee802-11'...
remote: Enumerating objects: 2773, done.
remote: Counting objects: 100% (29/29), done.
remote: Compressing objects: 100% (22/22), done.
remote: Total 2773 (delta 11), reused 14 (delta 7), pack-reused 2744
Réception d'objets: 100% (2773/2773), 1.03 Mio | 2.88 Mio/s, fait.
Résolution des deltas: 100% (1908/1908), fait.
-- The CXX compiler identification is GNU 9.4.0
-- The C compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gmp'
--   Found gmp, version 6.2.0
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Using GMP.
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.71.0") found components: date_time program_options filesystem system regex thread unit_test_framework 
-- Found Volk: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.10") 
-- Checking for module 'sndfile'
--   Found sndfile, version 1.0.28
-- Found SNDFILE: /usr/lib/x86_64-linux-gnu/libsndfile.so  
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.8
-- Found FFTW3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- Looking for xmmintrin.h
-- Looking for xmmintrin.h - found
-- Using install prefix: /usr/local
-- Building for version: 86965650 / 1.1.0git
-- Found SWIG: /usr/bin/swig4.0 (found version "4.0.1") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.10") 
CMake Error at swig/CMakeLists.txt:34 (include):
  include could not find load file:

    GrSwig

CMake Error at swig/CMakeLists.txt:48 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

-- Configuring incomplete, errors occurred!
See also "/home/tim/prefix-3.9/src/gr-ieee802-11/build/CMakeFiles/CMakeOutput.log".
See also "/home/tim/prefix-3.9/src/gr-ieee802-11/build/CMakeFiles/CMakeError.log".

bastibl, 2022-09-20:
It looks like you are trying to compile the 3.8 oder 3.7 version of this module with the 3.9 version of GNU Radio, which fails.

franckstik, 2022-09-22:
it's ok, I was able to install correctly from the sources. Thank you

closed 2022-09-23 by bastibl