bastibl/gr-ieee802-11 #365: HackRF One

pococ31, 2022-10-25:
I replaced the UHD Source with _"Soapy HackRF Source"_  and run the code in GNU radio companion version 3.10 but I am constantly getting "s00ss00ss00ss00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s00s" and the wireshark can't able to form any pcap from /tmp/wifi.pcap.

how to fix this issue ?

bastibl, 2022-10-25:
The HackRF has an uncompensated DC offset, which continuously triggers frame detection. This was discussed in many issues.

pococ31, 2022-10-25:
Thank you bastibl. Can you share the link of the discussed issues/suggestions that I could follow up.