bastibl/gr-ieee802-11 #367: Build failure with fmt > 8

kiufta, 2022-12-15:
spdlog 1.11.0 uses fmt 9.1.0, and this seems to incompatible:
```
In file included from /usr/include/spdlog/fmt/fmt.h:31,
                 from /usr/include/spdlog/common.h:45,
                 from /usr/include/gnuradio/logger.h:36,
                 from /usr/include/gnuradio/custom_lock.h:15,
                 from /usr/include/gnuradio/buffer.h:15,
                 from /usr/include/gnuradio/buffer_double_mapped.h:15,
                 from /usr/include/gnuradio/io_signature.h:23,
                 from /usr/include/gnuradio/basic_block.h:15,
                 from /usr/include/gnuradio/block.h:17,
                 from ./gr-ieee802-11/lib/../include/ieee802_11/mapper.h:20,
                 from ./gr-ieee802-11/lib/utils.h:22,
                 from ./gr-ieee802-11/lib/parse_mac.cc:17:
/usr/include/fmt/core.h: In instantiation of 'constexpr fmt::v9::detail::value<Context> fmt::v9::detail::make_value(T&&) [with Context = fmt::v9::basic_format_context<fmt::v9::appender, char>; T = const boost::basic_format<char>&]':
/usr/include/fmt/core.h:1777:29:   required from 'constexpr fmt::v9::detail::value<Context> fmt::v9::detail::make_arg(T&&) [with bool IS_PACKED = true; Context = fmt::v9::basic_format_context<fmt::v9::appender, char>; type <anonymous> = fmt::v9::detail::type::custom_type; T = const boost::basic_format<char>&; typename std::enable_if<IS_PACKED, int>::type <anonymous> = 0]'
/usr/include/fmt/core.h:1901:77:   required from 'constexpr fmt::v9::format_arg_store<Context, Args>::format_arg_store(T&& ...) [with T = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&}; Context = fmt::v9::basic_format_context<fmt::v9::appender, char>; Args = {boost::basic_format<char, std::char_traits<char>, std::allocator<char> >}]'
/usr/include/fmt/core.h:1918:31:   required from 'constexpr fmt::v9::format_arg_store<Context, typename std::remove_cv<typename std::remove_reference<Args>::type>::type ...> fmt::v9::make_format_args(Args&& ...) [with Context = basic_format_context<appender, char>; Args = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&}]'
/usr/include/spdlog/logger.h:374:75:   required from 'void spdlog::logger::log_(spdlog::source_loc, spdlog::level::level_enum, spdlog::string_view_t, Args&& ...) [with Args = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&}; spdlog::string_view_t = fmt::v9::basic_string_view<char>]'
/usr/include/spdlog/logger.h:90:13:   required from 'void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, fmt::v9::format_string<T ...>, Args&& ...) [with Args = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&}; fmt::v9::format_string<T ...> = fmt::v9::basic_format_string<char, const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&>]'
/usr/include/spdlog/logger.h:109:12:   required from 'void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, const T&) [with T = boost::basic_format<char>; typename std::enable_if<(! spdlog::is_convertible_to_any_format_string<const T&>::value), int>::type <anonymous> = 0]'
/usr/include/spdlog/logger.h:102:12:   required from 'void spdlog::logger::log(spdlog::level::level_enum, const T&) [with T = boost::basic_format<char>]'
/usr/include/spdlog/logger.h:279:12:   required from 'void spdlog::logger::info(const T&) [with T = boost::basic_format<char>]'
./gr-ieee802-11/lib/parse_mac.cc:67:9:   required from here
/usr/include/fmt/core.h:1757:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
 1757 |       formattable,
      |       ^~~~~~~~~~~
/usr/include/fmt/core.h:1757:7: note: 'formattable' evaluates to false
[ 25%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/utils.cc.o
In file included from /usr/include/spdlog/fmt/fmt.h:31,
                 from /usr/include/spdlog/common.h:45,
                 from /usr/include/gnuradio/logger.h:36,
                 from /usr/include/gnuradio/custom_lock.h:15,
                 from /usr/include/gnuradio/buffer.h:15,
                 from /usr/include/gnuradio/buffer_double_mapped.h:15,
                 from /usr/include/gnuradio/io_signature.h:23,
                 from /usr/include/gnuradio/basic_block.h:15,
                 from /usr/include/gnuradio/block.h:17,
                 from ./gr-ieee802-11/lib/../include/ieee802_11/decode_mac.h:20,
                 from ./gr-ieee802-11/lib/decode_mac.cc:17:
/usr/include/fmt/core.h: In instantiation of 'constexpr fmt::v9::detail::value<Context> fmt::v9::detail::make_value(T&&) [with Context = fmt::v9::basic_format_context<fmt::v9::appender, char>; T = const boost::basic_format<char>&]':
/usr/include/fmt/core.h:1777:29:   required from 'constexpr fmt::v9::detail::value<Context> fmt::v9::detail::make_arg(T&&) [with bool IS_PACKED = true; Context = fmt::v9::basic_format_context<fmt::v9::appender, char>; type <anonymous> = fmt::v9::detail::type::custom_type; T = const boost::basic_format<char>&; typename std::enable_if<IS_PACKED, int>::type <anonymous> = 0]'
/usr/include/fmt/core.h:1901:77:   required from 'constexpr fmt::v9::format_arg_store<Context, Args>::format_arg_store(T&& ...) [with T = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&}; Context = fmt::v9::basic_format_context<fmt::v9::appender, char>; Args = {boost::basic_format<char, std::char_traits<char>, std::allocator<char> >}]'
/usr/include/fmt/core.h:1918:31:   required from 'constexpr fmt::v9::format_arg_store<Context, typename std::remove_cv<typename std::remove_reference<Args>::type>::type ...> fmt::v9::make_format_args(Args&& ...) [with Context = basic_format_context<appender, char>; Args = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&}]'
/usr/include/spdlog/logger.h:374:75:   required from 'void spdlog::logger::log_(spdlog::source_loc, spdlog::level::level_enum, spdlog::string_view_t, Args&& ...) [with Args = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&}; spdlog::string_view_t = fmt::v9::basic_string_view<char>]'
/usr/include/spdlog/logger.h:90:13:   required from 'void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, fmt::v9::format_string<T ...>, Args&& ...) [with Args = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&}; fmt::v9::format_string<T ...> = fmt::v9::basic_format_string<char, const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&>]'
/usr/include/spdlog/logger.h:109:12:   required from 'void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, const T&) [with T = boost::basic_format<char>; typename std::enable_if<(! spdlog::is_convertible_to_any_format_string<const T&>::value), int>::type <anonymous> = 0]'
/usr/include/spdlog/logger.h:102:12:   required from 'void spdlog::logger::log(spdlog::level::level_enum, const T&) [with T = boost::basic_format<char>]'
/usr/include/spdlog/logger.h:279:12:   required from 'void spdlog::logger::info(const T&) [with T = boost::basic_format<char>]'
./gr-ieee802-11/lib/decode_mac.cc:150:9:   required from here
/usr/include/fmt/core.h:1757:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
 1757 |       formattable,
      |       ^~~~~~~~~~~
/usr/include/fmt/core.h:1757:7: note: 'formattable' evaluates to false
In file included from /usr/include/spdlog/fmt/fmt.h:31,
                 from /usr/include/spdlog/common.h:45,
                 from /usr/include/gnuradio/logger.h:36,
                 from /usr/include/gnuradio/custom_lock.h:15,
                 from /usr/include/gnuradio/buffer.h:15,
                 from /usr/include/gnuradio/buffer_double_mapped.h:15,
                 from /usr/include/gnuradio/io_signature.h:23,
                 from /usr/include/gnuradio/basic_block.h:15,
                 from /usr/include/gnuradio/block.h:17,
                 from ./gr-ieee802-11/lib/../include/ieee802_11/mapper.h:20,
                 from ./gr-ieee802-11/lib/utils.h:22,
                 from ./gr-ieee802-11/lib/sync_short.cc:17:
[... cut ...]
/usr/include/spdlog/logger.h:279:12:   required from 'void spdlog::logger::info(const T&) [with T = boost::basic_format<char>]'
./gr-ieee802-11/lib/sync_long.cc:116:21:   required from here
/usr/include/fmt/core.h:1757:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
 1757 |       formattable,
      |       ^~~~~~~~~~~
/usr/include/fmt/core.h:1757:7: note: 'formattable' evaluates to false
```

bastibl, 2022-12-16:
This is a GNU Radio issue. This project does not directly use fmt or spdlog.

kiufta, 2022-12-16:
I'm sorry!

closed 2022-12-18 by bastibl

kiufta, 2022-12-18:
I've reopened the issue because I may compile gnuradio and other out-of-tree modules successfully with the very same config.

Would it cost you much effort to try a build with spdlog 1.11.0 & fmt 9.1.0?

bastibl, 2022-12-18:
Please look at the output of the error message. This module just includes `block.h`. Also, you can search for `fmt` or `spdlog` in this module. AFAIS, they are not used. I'm not sure what I could change. Please check your local installation, in particular, if you have artifacts from old attempts somewhere.

kiufta, 2022-12-18:
I've just deleted my ccache, but the result stays the same.

Is frame_equalizer_impl.cc maybe using /usr/include/gnuradio/logger.h in a different manner than other modules?
```
    mylog(boost::format("encoding: %1% - length: %2% - symbols: %3%") % d_frame_encoding %
          d_frame_bytes % d_frame_symbols);
```

bastibl, 2022-12-19:
It's this commit: https://github.com/gnuradio/gnuradio/commit/4cd04154aaaa19419ec3410f024d6e935e0f7eb9 which is only in `main`, i.e., the dev branch. The `maint-3.10` branch of this module is supposed to be used with the `maint-3.10` branch of GNU Radio. So you either have to switch to `maint-3.10` or update the logger. They support format strings directly, i.e., boost::format is not necessary.

kiufta, 2022-12-19:
I'm sorry for being dumb. This is how the tree looks like:
* https://github.com/bastibl/gr-ieee802-11/blob/maint-3.10/lib/decode_mac.cc#L150
* https://github.com/bastibl/gr-ieee802-11/blob/maint-3.10/lib/frame_equalizer_impl.cc#L363

bastibl, 2022-12-19:
???

bastibl, 2022-12-19:
Nochmal: GNU Radio hat einen `main` branch der für die Entwicklung genutzt wird. Er ist nicht identisch mit `maint-3.10`. Er enthält das oben verlinkte Commit und ist deswegen nicht kompatibel mit dem Modul. Wie gehabt: der `maint-3.10`-Branch von diesem Modul ist für die Nutzung mit dem `maint-3.10`-Branch von GNU Radio vorgesehen...