bastibl/gr-ieee802-11 #369: Version number error with MacOS Ventura 13.2.1

shen-tianji-gn, 2023-03-08:
Error while building the package.

OS: MacOS 13.2.1 Ventura
Xcode: Ver 14.2
```
[  3%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/chunks_to_symbols_impl.cc.o
[  7%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/constellations_impl.cc.o
[ 10%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/decode_mac.cc.o
[ 14%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/equalizer/base.cc.o
[ 17%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/equalizer/comb.cc.o
[ 21%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/equalizer/ls.cc.o
[ 25%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/equalizer/lms.cc.o
[ 28%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/equalizer/sta.cc.o
[ 32%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/ether_encap_impl.cc.o
[ 35%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/extract_csi_impl.cc.o
[ 39%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/frame_equalizer_impl.cc.o
[ 42%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/mac.cc.o
[ 46%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/mapper.cc.o
[ 50%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/parse_mac.cc.o
[ 53%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/signal_field_impl.cc.o
[ 57%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/sync_long.cc.o
[ 60%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/sync_short.cc.o
[ 64%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/utils.cc.o
[ 67%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/viterbi_decoder/base.cc.o
[ 71%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/viterbi_decoder/viterbi_decoder_generic.cc.o
[ 75%] Linking CXX shared library libgnuradio-ieee802_11.dylib
ld: malformed 64-bit a.b.c.d.e version number: 86965650.0.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/libgnuradio-ieee802_11.86965650.dylib] Error 1
make[1]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/all] Error 2
make: *** [all] Error 2
```

shen-tianji-gn, 2023-03-08:
I hold arm64 and x64 devices. Both of them are in Ventura 13.2.1, and the same issues occur while `make`.

bastibl, 2023-03-08:
I don't have mac and, therefore, cannot reproduce the issue. Can you build any other GNU Radio module. Can you point me to one that works for you, then we can compare the cmake versioning stuff.

closed 2023-03-30 by bastibl