bastibl/gr-ieee802-11 #370: Section Explaination

thayheng, 2023-03-28:
Dear Sir,
I am quite new on DSP as well as gnu radio and, I am hoping you could help with some explanation of this part of gr_wifi_receiver. I am not sure if these part include 3 main things as 1. de-noising 2. time-recovery (clock sync) 3. channel equalization (multi-path effect). If so, where does these processes take place in the receiver?

[image]

thayheng, 2023-03-28:
Thank you sir for your response! For de-noising, I refer to filtering process for should there be any noise in the signal.

bastibl, 2023-03-29:
We don't do de-noising. I'm not sure what you had in mind specifically. But since the signal is not oversampled, there is no channel filtering or something like that.

thayheng, 2023-03-29:
Ah, I see. Thank you sir.

closed 2023-03-29 by thayheng