bastibl/gr-ieee802-11 #375: performance issue on mcs 6 (rate 48Mbps)

cloud9477, 2023-05-03:
Hi, I did a simple test on performance (packet delivery ratio vs SNR), the curve of mcs 6 (legacy rate 48 mbps) is weird, I was using the equalizer 0, attached please find my results:

psdu len 114:
[image]

psdu len 264:
[image]

psdu len 564:
[image]

bastibl, 2023-05-05:
did you run this with the most recent maint-3.10 branch?

cloud9477, 2023-05-05:
yes

bastibl, 2023-05-08:
Since this is the only MCS with a code rate of 2/3, I'd assume that there is an issue with certain frame sizes for this MCS. You could (1) use the generic decoder and see if it shows the same behavior or (2) look into rate-specific parts of the code, e.g., https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/viterbi_decoder/base.cc#L39-L59

bastibl, 2023-05-10:
Did you find the reason for the strange behavior?