bastibl/gr-ieee802-11 #377: bit string before channel encoding extraction

Acoustical, 2023-05-16:
Hi there! Thank you for providing us with this wonderful project. I have recently been studying this project and trying to output some bit strings. The specific bit strings I need fall into the following four types:

1. In the transmitter, the bit string before channel encoding (after source encoding).
2. In the transmitter, the bit string after channel encoding.
3. In the receiver, the bit string after channel decoding (before source decoding).
4. In the receiver, the bit string before channel decoding.

I have been able to easily obtain the bit strings for types 2 and 4 by adding a "File Sink" in GNU Radio. However, I believe I will need to modify the source code if I want to obtain the bit strings for types 1 and 3.

For type 1, I have found the channel encoding source code here: [mapper.cc_(line_116)](https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/mapper.cc?plain=1#L116). So, essentially, I just need to output the "scrambled_data," 

Regarding type 3, I have not yet found the source code. Could you please point me in the right direction and let me know which part of the code I should look for if I want to obtain the bit string after channel decoding?

Thank you so much!

bastibl, 2023-05-17:
See here: https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/decode_mac.cc#L136

Acoustical, 2023-05-17:
> See here: https://github.com/bastibl/gr-ieee802-11/blob/maint-3.8/lib/decode_mac.cc#L136

Thanks a lot! That's exactly what I was looking for!

closed 2023-05-17 by Acoustical