bastibl/gr-ieee802-11 #379: NameError: name 'ieee802_11' is not defined, when loading wifi_loopback.grc

ptef, 2023-06-09:
Hallo there.

I know this is a very annoying error and I'm sorry for opening issue about that. I tried everything I could think of.

The error I'm getting is this one:
```
$ gnuradio-companion /home/dukpt/Downloads/gr-ieee802-11/examples/wifi_loopback.grc 
>>> Warning: vocoder_codec2_decode_ps - option_attributes are for enums only, ignoring
>>> Warning: vocoder_codec2_encode_sp - option_attributes are for enums only, ignoring
<<< Welcome to GNU Radio Companion 3.10.5.1 >>>

Block paths:
	/usr/share/gnuradio/grc/blocks

Loading: "/home/dukpt/Downloads/gr-ieee802-11/examples/wifi_loopback.grc"
>>> Loading: /home/dukpt/Downloads/gr-ieee802-11/examples/wifi_phy_hier.grc
[ERROR] Failed to evaluate parameter block chan_est (FlowGraph.py:275)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 271, in renew_namespace
    value = eval(
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
[ERROR] Failed to evaluate parameter block encoding (FlowGraph.py:275)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 271, in renew_namespace
    value = eval(
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
[ERROR] Failed to evaluate parameter block chan_est (FlowGraph.py:275)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 271, in renew_namespace
    value = eval(
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
[ERROR] Failed to evaluate parameter block encoding (FlowGraph.py:275)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 271, in renew_namespace
    value = eval(
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
[ERROR] Failed to evaluate parameter block chan_est (FlowGraph.py:275)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 271, in renew_namespace
    value = eval(
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
[ERROR] Failed to evaluate parameter block encoding (FlowGraph.py:275)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 271, in renew_namespace
    value = eval(
  File "<string>", line 1, in <module>
NameError: name 'ieee802_11' is not defined
>>> Load Error: /home/dukpt/Downloads/gr-ieee802-11/examples/wifi_phy_hier.grc: Flowgraph invalid
**************************************************
6 errors from flowgraph:
Param - Value(value):
	Value "ieee802_11.signal_field()" cannot be evaluated:
	name 'ieee802_11' is not definedParam - Value(value):
	Value "ieee802_11.LS" cannot be evaluated:
	name 'ieee802_11' is not definedParam - Formatter Object(header_formatter):
	Value "header_formatter.formatter()" cannot be evaluated:
	'NoneType' object has no attribute 'formatter'Param - Value(value):
	Value "ieee802_11.BPSK_1_2" cannot be evaluated:
	name 'ieee802_11' is not definedParam - Algorithm(algo):
	Value "chan_est" cannot be evaluated:
	name 'chan_est' is not definedParam - Encoding(encoding):
	Value "encoding" cannot be evaluated:
	name 'encoding' is not defined
**************************************************
>>> Done
```
The gnuradio-companion print:
[image]

GNU Radio was installed from the main repository, following [Installing instructions in GNU Radio website](https://wiki.gnuradio.org/index.php/InstallingGR) for Ubuntu 22.04.

```
$ gnuradio-config-info --prefix
/usr/
$ gnuradio-config-info --sysconfdir
/etc
$ gnuradio-config-info --print-all
/usr/
/etc
/etc/gnuradio/conf.d
/home/dukpt/.gnuradio
Mon, 06 Feb 2023 11:23:06Z
testing-support;python-support;post-install;doxygen;man-pages;gnuradio-runtime;common-precompiled-headers;gr-ctrlport;* thrift;gnuradio-companion;JSON/YAML config blocks;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-audio;* alsa;* oss;* jack;* portaudio;gr-channels;gr-pdu;gr-iio;* libad9361;gr-qtgui;gr-trellis;gr-uhd;gr-uhd UHD 4.0 RFNoC;gr-utils;gr_modtool;gr_blocktool;gr-video-sdl;gr-vocoder;* codec2;* freedv;* gsm;gr-wavelet;gr-zeromq;gr-network;gr-soapy
3.10.5.1
cc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0 
Copyright (C) 2021 Free Software Foundation, Inc. 
This is free software see the source for copying conditions.  There is NO 
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
c++ (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0 
Copyright (C) 2021 Free Software Foundation, Inc. 
This is free software see the source for copying conditions.  There is NO 
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
/usr/bin/cc:::-O3 -DNDEBUG -g -O2 -ffile-prefix-map=/build/gnuradio-8rZ5NI/gnuradio-3.10.5.1-0~gnuradio~jammy=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized 
/usr/bin/c++:::-O3 -DNDEBUG -g -O2 -ffile-prefix-map=/build/gnuradio-8rZ5NI/gnuradio-3.10.5.1-0~gnuradio~jammy=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
2.9.1
```

I've cloned from the branch maint-3.10 as I saw you pointed out in some other answer to work with GNU Radio 3.10:   
`$ git clone https://github.com/bastibl/gr-ieee802-11 --branch maint-3.10
`

Also python is able to correctly import module (I use [asdf](https://asdf-vm.com/) as version manager):
```
$ python
Python 3.11.3 (main, May  9 2023, 16:11:59) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ieee802_11
>>> 
$ which python
/home/dukpt/.asdf/shims/python
```

I've already checked issue #268, #208, #192 but none was enough. Somehow python inside GNU Radio is not able to find ieee802_11 module, even after specifying `PYTHONPATH` and `LD_LIBRARY_PATH` as I saw in other issues.
```
$ echo $PYTHONPATH 
/usr/lib/python3/dist-packages:/home/dukpt/.local/lib/python3.11/site-packages
dukpt@PROMETHEUS:~$ echo $LD_LIBRARY_PATH 
/home/dukpt/.local/lib:/usr/lib:/usr/local/lib
dukpt@PROMETHEUS:~$ ls -ld /lib
lrwxrwxrwx 1 root root 7 Mai  7 13:54 /lib -> usr/lib
$ find /usr -type d -name ieee802_11
/usr/include/ieee802_11
/usr/lib/cmake/ieee802_11
/usr/lib/python3/dist-packages/ieee802_11
$ find /usr -type d -name gnuradio
/usr/share/gnuradio
/usr/share/gnuradio/modtool/templates/gr-newmod/include/gnuradio
/usr/share/doc/gnuradio
/usr/include/gnuradio
/usr/lib/x86_64-linux-gnu/cmake/gnuradio
/usr/lib/python3/dist-packages/gnuradio
$ find /usr -type d -name foo
/usr/local/lib/python3.10/dist-packages/foo
/usr/local/lib/cmake/foo
/usr/local/include/foo
/usr/include/foo
/usr/lib/cmake/foo
/usr/lib/python3/dist-packages/foo
```

Considering what you've pointed out in previous answers:
- _incompatible versions of GNU Radio and this module_ - I think this was addressed by cloning the correct branch mentioned by you;
- _the install path of the Python module was not in the PYTHONPATH_ - done;
- _the install path of the shared library was not in LD_LIBRARY_PATH_ - also done;
- _the shared library was installed in a system directory and sudo ldconfig was not run_ - initially I had done this wrong and I thought it would work after your suggestion, however the error persisted.
- gr-foo installed in the wrong directory - fixed

Is it possible to have some differences between `main` und `maint-3.10` branches that could cause problems?  
Or perhaps I'm still missing something?

Thank you.

ptef, 2023-06-12:
It's not possible to build because the gadgets are not found.   

Anyway I reconfigured distro python and removed the local python managed by asdf (I think it was conflicting since gnuradio use /usr/bin/python3):
`$ sudo apt install --reinstall python3 python3-minimal --fix-broken
`   

But now I have another problem that maybe is easier to catch:

```
$ python3 
Python 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gnuradio import gr
>>> import ieee802_11
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/ieee802_11/__init__.py", line 18, in <module>
    from .ieee802_11_python import *
ImportError: generic_type: type "chunks_to_symbols" referenced unknown base type "gr::tagged_stream_block"
>>> import foo
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/foo/__init__.py", line 18, in <module>
    from .foo_python import *
ImportError: generic_type: type "burst_tagger" referenced unknown base type "gr::sync_block"
>>> 
```

Both gr-foo and gr-ieee802_11 were configured with `cmake ../ -DCMAKE_INSTALL_PREFIX=/usr`

The environment is also configured:

```
$ echo $LD_LIBRARY_PATH 
/lib:/usr/lib:/usr/local/lib:/home/dukpt/.local/lib
$ echo $PYTHONPATH 
/usr/lib/python3/dist-packages:/usr/lib/python3.10/site-packages
$ find /usr -type d -name ieee802_11
/usr/include/ieee802_11
/usr/lib/cmake/ieee802_11
/usr/lib/python3.10/site-packages/ieee802_11
$ find /usr -type d -name foo
/usr/local/include/foo
/usr/include/foo
/usr/lib/cmake/foo
/usr/lib/python3.10/site-packages/foo
$ find /usr -type d -name gnuradio
/usr/share/gnuradio
/usr/share/gnuradio/modtool/templates/gr-newmod/include/gnuradio
/usr/share/doc/gnuradio
/usr/include/gnuradio
/usr/lib/x86_64-linux-gnu/cmake/gnuradio
/usr/lib/python3/dist-packages/gnuradio
```

Any ideas?

bastibl, 2023-06-14:
That looks like a PyBind version conflict. Can you please try the one that comes with your distro.

ptef, 2023-06-19:
This is done already, but not working:   

```
$ find /usr -type d -name pybind11
/usr/include/pybind11
/usr/lib/cmake/pybind11
/usr/lib/python3/dist-packages/pybind11
/usr/lib/python3/dist-packages/pybind11/share/cmake/pybind11
/usr/lib/python3/dist-packages/pybind11/include/pybind11
```

```
$ python3
Python 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gnuradio import gr
>>> import pybind11
>>> import foo
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.10/site-packages/foo/__init__.py", line 18, in <module>
    from .foo_python import *
ImportError: generic_type: type "burst_tagger" referenced unknown base type "gr::sync_block"
>>> import ieee802_11
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.10/site-packages/ieee802_11/__init__.py", line 18, in <module>
    from .ieee802_11_python import *
ImportError: generic_type: type "chunks_to_symbols" referenced unknown base type "gr::tagged_stream_block"
>>> 
```

```
$ echo $PYTHONPATH 
/usr/lib/python3/dist-packages:/usr/lib/python3.10/site-packages:/home/dukpt/.local/lib/python3.10/site-packages
$ echo $LD_LIBRARY_PATH 
/lib:/usr/lib:/usr/local/lib:/home/dukpt/.local/lib

```

Both `gr-foo` and `gr-ieee802_11` were compiled with `-DCMAKE_INSTALL_PREFIX=/usr`

bastibl, 2023-06-29:
Again, your output shows that you have pybind from your package manager installed. It does not show that there is not other version from asdf (or whatever you used) or pip or virtualenv or ... I'm pretty sure either GNU Radio or the OOT or both were not compiled with the correct pybind version.

ptef, 2023-06-29:
Regarding pybind I already showed that it was installed from package manager and there is not other version installed:    
```
$ find ~/ /usr -type d -name pybind\*
/usr/share/doc/pybind11-dev
/usr/include/pybind11
/usr/lib/cmake/pybind11
/usr/lib/python3/dist-packages/pybind11-2.9.1.egg-info
/usr/lib/python3/dist-packages/pybind11
/usr/lib/python3/dist-packages/pybind11/share/cmake/pybind11
/usr/lib/python3/dist-packages/pybind11/include/pybind11
/usr/lib/python3/dist-packages/pybindgen

$ dpkg-query -L python3-pybind11
/.
/usr
/usr/bin
/usr/bin/pybind11-config
/usr/lib
/usr/lib/python3
/usr/lib/python3/dist-packages
/usr/lib/python3/dist-packages/pybind11
/usr/lib/python3/dist-packages/pybind11/__init__.py
/usr/lib/python3/dist-packages/pybind11/__main__.py
/usr/lib/python3/dist-packages/pybind11/_version.py
/usr/lib/python3/dist-packages/pybind11/_version.pyi
/usr/lib/python3/dist-packages/pybind11/commands.py
/usr/lib/python3/dist-packages/pybind11/include
/usr/lib/python3/dist-packages/pybind11/include/pybind11
/usr/lib/python3/dist-packages/pybind11/include/pybind11/attr.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/buffer_info.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/cast.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/chrono.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/common.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/complex.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/detail
/usr/lib/python3/dist-packages/pybind11/include/pybind11/detail/class.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/detail/common.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/detail/descr.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/detail/init.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/detail/internals.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/detail/type_caster_base.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/detail/typeid.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/eigen.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/embed.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/eval.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/functional.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/gil.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/iostream.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/numpy.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/operators.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/options.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/pybind11.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/pytypes.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/stl
/usr/lib/python3/dist-packages/pybind11/include/pybind11/stl/filesystem.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/stl.h
/usr/lib/python3/dist-packages/pybind11/include/pybind11/stl_bind.h
/usr/lib/python3/dist-packages/pybind11/py.typed
/usr/lib/python3/dist-packages/pybind11/setup_helpers.py
/usr/lib/python3/dist-packages/pybind11/setup_helpers.pyi
/usr/lib/python3/dist-packages/pybind11/share
/usr/lib/python3/dist-packages/pybind11/share/cmake
/usr/lib/python3/dist-packages/pybind11/share/cmake/pybind11
/usr/lib/python3/dist-packages/pybind11/share/cmake/pybind11/FindPythonLibsNew.cmake
/usr/lib/python3/dist-packages/pybind11/share/cmake/pybind11/pybind11Common.cmake
/usr/lib/python3/dist-packages/pybind11/share/cmake/pybind11/pybind11Config.cmake
/usr/lib/python3/dist-packages/pybind11/share/cmake/pybind11/pybind11ConfigVersion.cmake
/usr/lib/python3/dist-packages/pybind11/share/cmake/pybind11/pybind11NewTools.cmake
/usr/lib/python3/dist-packages/pybind11/share/cmake/pybind11/pybind11Targets.cmake
/usr/lib/python3/dist-packages/pybind11/share/cmake/pybind11/pybind11Tools.cmake
[... 140 lines cut ...]
-- Up-to-date: /usr/share/gnuradio/grc/blocks/ieee802_11_mapper.block.yml
-- Up-to-date: /usr/share/gnuradio/grc/blocks/ieee802_11_parse_mac.block.yml
-- Up-to-date: /usr/share/gnuradio/grc/blocks/ieee802_11_sync_long.block.yml
-- Up-to-date: /usr/share/gnuradio/grc/blocks/ieee802_11_sync_short.block.yml
```

Regarding the line
`-- Set runtime path of "/usr/lib/python3.10/site-packages/ieee802_11/ieee802_11_python.cpython-310-x86_64-linux-gnu.so" to ""` 
do you think this could indicate some reference problem?

```
$ python3 -c 'import foo' 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.10/site-packages/foo/__init__.py", line 18, in <module>
    from .foo_python import *
ImportError: generic_type: type "burst_tagger" referenced unknown base type "gr::sync_block"

$ python3 -c 'import ieee802_11' 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.10/site-packages/ieee802_11/__init__.py", line 18, in <module>
    from .ieee802_11_python import *
ImportError: generic_type: type "chunks_to_symbols" referenced unknown base type "gr::tagged_stream_block"
```

bastibl, 2023-06-29:
All I can say is that I had this exact problem and it turned out to be a pybind issue.

The exact same problem is also identified as a pybind issue here: https://lists.gnu.org/archive/html/discuss-gnuradio/2021-05/msg00056.html

Please try to compile any other GNU Radio module. If other modules work, let me know. Otherwise, please write to the GNU Radio mailing list.

closed 2023-08-17 by bastibl