bastibl/gr-ieee802-11 #380: burst power

mbalalzade, 2023-06-11:
hello there
i want to measure the burst power of wifi signal using this rep. and i want to know is it possible?

bastibl, 2023-06-12:
Measuring the burst power is problem for the hardware and its drivers. You need to check documentation of the vendor to see if this is a calibrated device that allows mapping the frequency, sample rate, bandwidth, and gain to absolute power values. Then you could use the amplitudes at various receiver steps and map them to absolute power values.

mbalalzade, 2023-06-12:
thanks for your help but now i want ask with rx project in this repository may i scan the wifi signal  or i have to use another sdr platform to run tx project and then with another sdr and rx project recive signal?

bastibl, 2023-06-12:
You can receive frames from your normal AP (i.e., you don't have to generate them with another SDR). Set the sample rate to 20MHz and choose the corresponding channel. It will only decode 802.11a/g/p. Easiest is to open a network in 5GHz band. On 2.4GHz, there may be strange compatibility modes (even though it usually works there as well).

mbalalzade, 2023-06-12:
thank you so much

mbalalzade, 2023-06-12:
Please accept my apology to ask another question. i want to know is it possible to plot spectrum of received signal. and i heard that power of burst signal is more than the general signal is it true?

mbalalzade, 2023-06-12:
for example in a place with 4 wifi access point i want to know when i receive signal how i can or how rx.grc can separate them from each other it means may i receive sum of 4 wifi signal?

closed 2023-06-17 by mbalalzade