bastibl/gr-ieee802-11 #385: X310 multi-channel transmission issue

GaryGogh, 2023-06-27:
Dear bastibl,

I hope this message finds you well. I am currently using the USRP X310 as my transmitting device and employing the monitoring mode of a commercial NIC on the receiving end. When I utilize only one channel of the X310, I can successfully observe signals and data packets from the USRP in Wireshark on the receiving side. However, when I attempt to utilize both channels of the X310 and modify the workflow as shown below, the receiving end no longer detects any signals. I have tried various combinations of gain and sampling rates, but none of them have been effective. I would greatly appreciate your insights on what could be causing this issue.
[image]

Thank you in advance for your assistance.

Best regards,
Gary

bastibl, 2023-06-28:
Such a flowgraph could cause problems when the number of samples per packet differ in the two copies of the flowgraph, since the USRP sink always consumes the same number of samples on each branch. In theory that should not be a problem in your setup, since it looks like the parameters are similar (I guess the packet size of the two message sources are similar). Still you could try
- connecting the same stream to the two ports
- connect a Null Source to one input
- send some test signals on the two ports to check if you device config is fine

Another issue could be that doubling the data rate to the device is the problem. See the UHD manual if there are some performance optimizations (like the samples per packet).

GaryGogh, 2023-07-06:
Dear bastibl, 
Thanks for your prompt and detailed reply. I have been conducting extensive tests this week, but unfortunately, I haven't been able to observe any content in Wireshark on the receiving end. However, during my experimentation, I made a few interesting discoveries that I would like to share and discuss with you. Additionally, I have a couple of questions that I would appreciate your insights on.

Before proceeding, I would like to emphasize that I have already verified the proper functioning of both channels of the USRP, indicating that there are no hardware issues with the devices.

Firstly, I came across your suggestion of setting the sampling rate to 20MHz when using a commercial network card for reception. However, I couldn't find any specific explanation for this recommendation. Furthermore, I am curious to know if the receiving end also requires the sampling rate to be set at 20MHz. It seems to me that a higher sampling rate on the receiving end might potentially yield a finer signal granularity. Could you kindly provide further clarification on the rationale behind this recommendation?
Secondly, when I configured both channels in Gnuradio with a sampling rate of 20MHz, I encountered overflow errors. Interestingly, even when I set one channel's source as the Null Source, the overflow error persisted. However, I am unsure if these overflow errors could cause the receiving end to completely fail in detecting any signal. Therefore, I would greatly appreciate your guidance in confirming whether my understanding is accurate or if there might be other factors at play.
I genuinely appreciate your expertise and assistance in addressing these queries and concerns. Thank you kindly for your time and support!
Yours,
Gary

bastibl, 2023-07-06:
1.) Because 11a/g use 20MHz channels. See the IEEE 802.11 standard or search on Google for secondary sources. You cannot just increase the sample rate (i.e., oversample the signal) without changing the receiver accordingly. It is designed to sample with the bandwidth of the signal.
2.) You shouldn't get overflow errors (which occur on RX side afaik) but under runs, i.e., the sample stream stalls as samples do not arrive fast enough at the device, which corrupts the signal that is sent. You could try on the UHD mailing list, since this is an issue with UHD multi-channel operation and I don't have this setup to test.

closed 2023-08-17 by bastibl