bastibl/gr-ieee802-11 #391: Overflow when using HackRF on wifi_rx.grc

CodeDragon82, 2023-08-12:
Overflow when using HackRF source on wifi_rx.grc:
[image]

(Same problem with both Soapy HackRF Source and osmocom Source.)

The only thing I've changed in the source input:
[image]

Extra info:
 - I have tested on a Ubuntu system, WSL and a Kali Linux VB with the same results.
 - Using the maint-3.10 branch.
 - Installed:
   - gr-foo
   - gr-ieee802-11
   - gr-osmocom
   - hackrf (via apt)
   - libhackrf-dev (via apt)

(Please let me know if I'm missing any information that will be helpful.)

razgino11, 2023-08-14:
I have encountered the same issue exactly, will update here if i figure it out, keep me updated if you do as well :)

bastibl, 2023-08-15:
The HackRF has an uncompensated DC offset, which constantly triggers frame detection, overloading your PC. This was answered several times. Please search for HackRF in the Issues. In short, you somehow have to remove the DC spike through a rather sharp filter. Or 3rd party DC blocker blocks. Maybe the osmocom blocks can also do it.

razgino11, 2023-08-16:
@bastibl Thank you very much! setting it to 20 MHz and selecting the correct frequencies worked for me and im getting packets now using hackrf :)

CodeDragon82, 2023-08-16:
@bastibl Thank you! I removed the DC spike and I'm getting WiFi frame in Wireshark. I've encountered another problem though. The algorithm seems to only pick up on short frames such as ACKs, CTSs and deauths, and not long frames such as data frames and beacons. Any idea on why this might be?

razgino11, 2023-08-16:
@CodeDragon82 
Checked on my setup and i have the same issue like you again, hadn't noticed i was only getting those packets before

closed 2023-09-16 by bastibl

Cobi-Coder, 2025-03-25:
[image]
I used the CorrectIQ block, but the 'O' overflow still occurs. What should I do?

bastibl, 2025-03-25:
Either your PC is just not fast enough and cannot process the samples in real-time. Or the DC offset compensation is not good enough and it still triggers frame detection the whole time. One option is to run htop (with thread names) to get an idea whats going on. Or you use a tag debug after the sync short block to check if frame detection is triggered permanently.

Cobi-Coder, 2025-03-31:
[image]

[image]

Cobi-Coder, 2025-03-31:
Asus TUF FA507 ( R7 7435HS / Ram 8 / SSD 512 / RTX 4060 / Màn 15,6 inch FHD 144hz 100% sRGB )
May be I will buy this, do you think it will work?

bastibl, 2025-03-31:
In case you run this as a VM, try it natively. An i7 should be good enough.

It seems you are missing some basics. Maybe you should start with some of the examples and tutorials that GNU Radio provides: https://wiki.gnuradio.org/index.php/Tutorials