bastibl/gr-ieee802-11 #392: GrSwig Missing

smsajjadi, 2023-08-14:
Hi

On ubuntu 22.4, I have installed gnu-radio through pybombs. The installed version is 3.10.7.0. 
When I try to add gr-ieee802-11 to this installation, the following error is issued:
```

CMake Error at swig/CMakeLists.txt:34 (include):
  include could not find requested file:

    GrSwig

CMake Error at swig/CMakeLists.txt:48 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".
```

I think that this is due to the transition from swig to pybind.

Might there be an update so that this module could be used on newer versions of Gnu Radio?

smsajjadi, 2023-08-16:
I think yes. I did specify: "On ubuntu 22.4, I have installed gnu-radio through pybombs. The installed version is 3.10.7.0.". Is that ambiguous?

Anyhow this is gnuradio-config-info --printall results:

```
testing-support;python-support;post-install;doxygen;man-pages;gnuradio-runtime;common-precompiled-headers;gr-ctrlport;* thrift;gnuradio-companion;JSON/YAML config blocks;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-audio;* alsa;* oss;* jack;* portaudio;gr-channels;gr-pdu;gr-iio;* libad9361;gr-qtgui;gr-trellis;gr-uhd;gr-uhd UHD 4.0 RFNoC;gr-utils;gr_modtool;gr_blocktool;gr-video-sdl;gr-vocoder;* codec2;* freedv;* gsm;gr-wavelet;gr-zeromq;gr-network;gr-soapy
3.10.7.0
cc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0 
Copyright (C) 2021 Free Software Foundation, Inc. 
This is free software see the source for copying conditions.  There is NO 
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
c++ (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0 
Copyright (C) 2021 Free Software Foundation, Inc. 
This is free software see the source for copying conditions.  There is NO 
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
/usr/bin/cc:::-O3 -DNDEBUG -g -O2 -ffile-prefix-map=/build/gnuradio-gSVqHA/gnuradio-3.10.7.0-0~gnuradio~jammy=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized 
/usr/bin/c++:::-O3 -DNDEBUG -g -O2 -ffile-prefix-map=/build/gnuradio-gSVqHA/gnuradio-3.10.7.0-0~gnuradio~jammy=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
2.9.1

```

razgino11, 2023-08-16:
I might be misunderstanding you, but what i asked is if you are using this repo after you did a checkout to the "maint-3.10" branch, which has the code to support grc version 3.10

smsajjadi, 2023-08-16:
Oh. I misundestood you. Sorry. I thaught the master branch must be supporting the latest version of gnu radio, and did not note the various branches out there. Really sorry.

khen-su, 2023-09-08:
I have got the same issue though. Except I am using gnuradio 3.8.2.0. I have fetched maint-3.8 and I am now stuck with that missing GrSwig on cmake :/ Should I somehow install manually GrSwig.cmake?

sarielkk, 2023-09-15:
> Hi
>
> On ubuntu 22.4, I have installed gnu-radio through pybombs. The installed version is 3.10.7.0. When I try to add gr-ieee802-11 to this installation, the following error is issued:
>
> ```
>
> CMake [...]

Hello，did you solve the problem?

bastibl, 2023-09-15:
Again, this is a version issue. Citing from the readme: 

> Like GNU Radio, this module uses maint branches for development. These branches are supposed to be used with the corresponding GNU Radio branches. This means: the maint-3.7 branch is compatible with GNU Radio 3.7, maint-3.8 is compatible with GNU Radio 3.8, etc.

closed 2023-09-16 by bastibl